sanni/cartreader #725: Problem with OLED display

luk3ros, 2023-03-05:
Hello,
I have just build Card Reader Version 3 few minutes ago. I have problem with screen that displays on OLED. It looks like not all horizontal lines are drawn. The bottom of the screen is not affected. Everything is fine there. I have shown it in the photo. I resoldered the connections from both sides of the PCB again, but it didn't change anything. Is the screen damaged and needs to be replaced? Please help. 
[image]

sanni, 2023-03-05:
I had a case like this once, buying a new OLED fixed it.

luk3ros, 2023-03-06:
Thank you for your answer. I ordered a new screen today. I'll let you know if the problem is solved. Regards.