sanni/cartreader #734: SNES: Issue dumping PGA Tour 96

ValidusRazgriz, 2023-03-10:
I'm having an issue dumping the SA1 chip game PGA Tour 96 for the SNES. I'm using a HW5 Rev3 cart reader with V12.3 firmware and I can dump Super Mario RPG, Kirby Super Star, and Kirby's Dream Land 3 without any issues. When I try to dump PGA Tour 96 it gives me this error:

ERROR
Rom header corrupt
or missing

Press button for
manual configuration
or powercycle if SA1

I get the same error with two different PGA Tour 96 cartridges, both of which play fine on an SNES.  One of the cartridge PCB's is SHVC-1L3B-01 and the other is SHVC-1L3B-02.

nsx0r, 2023-03-17:
Hi, I have a similar issue with the PAL version of PGA Tour 96 (PCB = SNSP-1L3B-01)

[image]

sanni, 2023-03-18:
skaman and myself have noticed in the past that different SA1 games behave differently. Most likely there are different revisions of the SA1 chip. Some are way harder to unlock than others. SA1 was marketed as a copy protection if I remember correctly so it would make sense if they had updated the chips during its lifecycle, but that is just speculations on my part.

Personally I think that the current unlocking code for SA1 is flawed since the clock of the Arduino Mega and the clocks of the Adafruit Clock Generator are not in sync like they would be on a real SNES. 

There is also a second thing I have noticed but not have investigated further: I had an old Super Famicom that still had the 4Mhz CIC clock, and on that Super Famicom I could not unlock SA1 games with a PIC with the snesCIC hexfile, only with a PIC with the superCIC hex file. 
I made the CIC part of HW5 modular because of that, so I can test the snesCIC, superCIC and a real SNES CIC to see if there are differences. But I didn't have to time to even order the CIC PCBs yet.

Unlocking SA1 games is certainly something that would need further investigation using a logic analyzer. So far I have not checked any timings of any system with a logic analyzer, other than N64 Eeprom/Controller Pak. All timings are just best guesses. This is something I always wanted to check and fine-tune. 

The first thing would definitely be to understand why exactly the unlocking fails. Like capture the CIC communication of a successful and an unsuccessful unlock and then compare them, see where they differ.

Ancyker, 2023-03-21:
I ordered a US version, will test it, and see if I can reproduce the issue.

[image]

ValidusRazgriz, 2023-03-21:
I'm planning on testing out PGA European Tour when I get a hold of a copy. The only other North American SA1 game is Power Rangers Zeo: Battle Racers which I may also try to get.

nsx0r, 2023-03-22:
I can confirm the problem comes from the PIC12F629 / snescic-lock-resync.hex.
I replaced it with a PIC16F630 / supercic-lock_p16F630.hex and the dump is good:
[image]

Ancyker, 2023-03-23:
Just came to post that I could not reproduce it with a US cartridge...

[image]

nsx0r, 2023-03-24:
> Just came to post that I could not reproduce it with a US cartridge...
> 
> [image]

Are you using a snesCIC or a SuperCIC chip?

sanni, 2023-03-24:
PAL with PIC12F629/snescic-lock-resync works here too:

[image]

[image]

Ancyker, 2023-03-24:
> Are you using a snesCIC or a SuperCIC chip?

PIC12F629

Ancyker, 2023-03-24:
> I'm planning on testing out PGA European Tour when I get a hold of a copy. The only other North American SA1 game is Power Rangers Zeo: Battle Racers which I may also try to get.

I have that game; it's what I use to test SA-1 with, lol.

[image]
[image]

sanni, 2023-03-24:
Panic Bomberworld is the only other SA1 game I have.

[image]

[image]

Both my SA1 cartridges can be quite temperamental. Like right now I can't seem to unlock PGA 96 anymore, always header error, so I try Bomberworld and it dumps fine on the first try, then I try PGA 96 again and it also dumps fine first try. Almost as if the SA1 locks up and then you have to wait for the caps to fully drain before trying again.

sakman55, 2023-07-04:
I dumped all the Japan and US SA-1 carts during my testing.  I do have one somewhat stubborn PGA Tour 96 (US) cart that sometimes doesn't unlock.  Several other copies of the cart work fine.

superbonaci, 2023-08-26:
I'm wondering if the hardware from this link is related to this project: [Open Source Cartridge Reader V3-ALTER Build service [Group F]](https://savethehero.builders/collections/build-service/products/open-source-cartridge-reader-v3-alter-build-service)

since it has a purchase option "SA-1 UNLOCK SNES & N64 EEPROM SAVE (OPTION FEE $28 [PIC12F629 MICROCONTROLLER WITH THE SNESCIC AND CLOCK GENERATOR])"

If I have such board borrowed from a friend, how can I tell if it was purchased with the SA-1 option or not? Any way to tell?

Ancyker, 2023-08-26:
@superbonaci 

I do not have their cart reader, but this is the CIC unlock chip:
[image]

And I believe this is right under here:
[image]

You might be able to see it from the back.

You can also look for the clock gen:
[image]

Since the board is blue it should stand out. It should be visible from the side with the switches.

You can also just try it and see. The worst that will happen is it won't work (the dump will fail). It won't damage anything.

superbonaci, 2023-09-04:
@Ancyker thanks a lot for the explanation!

petrutoader, 2023-09-20:
We've been developing a separate SNES cartridge reader for Epilogue (epilogue.co). We're currently facing the same issues as this project. Some cartridges unlock properly while some others don't.

Our project uses clock generators from crystals while from what I've seen this project uses a clock generator IC from Adafruit.

In the following days we'll be building a separate module with a FPGA (DE-10 Nano) to see if the unlocking mechanism works better when the CIC_CLK is modulated in a similar fashion as the original SNES.

We usually release our code closed source, but in this case if we manage to make it a separate module we're open to sharing the code so that it could be implemented for this project as well and benefit the community.

petrutoader, 2023-09-22:
A brief update, we've replicated the clock for MCK, SYS_CLK and CIC_CLK using a DE10-Lite and integrated it on our boards. We do not have the hardware for this project (sanni/cartreader) to try it as well there, but it did not improve the unlocking mechanism on our boards.

We didn't have an exact 21.477MHz crystal available to use on the FPGA and derive the clock signals from it, but we used a 50MHz clock and the PLL integrated, having extremely close results.

[image]

Even if the clock could be part of the problem, it's not the main culprit. We'll look into the boot sequence of the supercic project to see what the issue is. For the SA-1 carts we can't seem to unlock, mostly Japanese titles, we see no data on the CIC_DATA0 and CIC_DATA1 lines.

petrutoader, 2023-10-20:
We've successfully unlocked the SA-1 on our devices without needing the FPGA to modulate SYS_CLK. Both @sanni and @nsx0r have pointed out correctly that the PIC16F630 tends to have a higher success rate.

Statistics:

- We've conducted 5,000 unlock attempts (with header data reading and checksum validation).
- Our current failure rate is approximately <1%.
- On average, we are achieving 3 unlocks every second, although there are occasional instances where it drops to 1 unlock per second.
- Our goal is to bring the failure rate down to 0.1%. Once we achieve this, we'll provide the waveforms or Saleae dump files for further insights and to help improve this project as well.

Key Observations:

- The capacitance of the cartridge's power rail is crucial. If unlocks occur too rapidly without allowing the power rail to decrease below 100mV, the rate of unsuccessful unlocks increases.
- The timing of when each clock begins, in relation to the moment the PAIR pin goes low, is a critical factor to consider.
- It seems that we're using different frequencies compared to this project for the clock generation (at the moment 21.47MHz and 3.57MHz for SYS_CLK and 1.785MHz CIC_CLK).

Additional Note:

- We're currently acquiring various SA-1 PCBs to comprehensively test our code. Preliminary findings suggest that not all SA-1 PCBs possess identical electrical characteristics, which can influence the timings.
- We don't have the sanni device on our hands so maybe someone can try the frequencies I mentioned to see if it leads to better unlocking.

cmcm711, 2023-10-20:
> We don't have the sanni device on our hands so maybe someone can try the frequencies I mentioned to see if it leads to better unlocking.

I have a fully-built HW5 Rev3 I can send you for the cost of parts and shipping, if it will help with this. Send me a DM in the community Discord server.

Ancyker, 2023-10-20:
> The capacitance of the cartridge's power rail is crucial. If unlocks occur too rapidly without allowing the power rail to decrease below 100mV, the rate of unsuccessful unlocks increases. We've tested 110uF initially and had better success on 20uF on our board's side.

@petrutoader This is interesting. I was trying to make a new six-slot PCB with a bulk cap on it and was having issues, I didn't consider the bulk cap could be why, I thought it was some bad trace routing. Thanks for that.

P.S. I have your GB Operator :3 
[image]

petrutoader, 2023-10-24:
> > We don't have the sanni device on our hands so maybe someone can try the frequencies I mentioned to see if it leads to better unlocking.
> 
> I have a fully-built HW5 Rev3 I can send you for the cost of parts and shipping, if it will help with this. Send me a DM in the community Discord server.

We've ordered all PCBs needed since we had most of the hardware needed in our inventory already. Thanks you for your kind offer!

> @petrutoader This is interesting. I was trying to make a new six-slot PCB with a bulk cap on it and was having issues, I didn't consider the bulk cap could be why, I thought it was some bad trace [...]

Beautiful setup and thank you! The bulk cap was really annoying since it extends all timings by quite a lot, the time to discharge makes the unlock very slow. I'm continuing my work on unlock stability, once I get to a satisfactory result I'm looking forward to share what we've implemented.

superbonaci, 2023-12-09:
Had no idea that the cartridges were so finicky about the input voltage. What I've heard in some youtube video is that the pitch of the game sound is higher as the console temperature is higher, probably it increases the clock frequency slightly or some.

petrutoader, 2023-12-11:
This is a solid point you're making @superbonaci and it's something I hadn't considered before. It's possible that the success rate of unlocking is linked to the clock frequency (MCK & CIC_CLK) and the voltage applied to the cartridge: the two being in a relationship.

Currently, we are in the process of determining the optimal input voltage for the cartridge. I am conducting tests on around 18 different SA-1 games with varying PCB serial numbers. Notably, these PCB serials exhibit differences, ranging from capacitance in the I/O and power traces to potentially different batches of SA-1 chips.

I plan to provide a report in a comment. Although our testing is conducted with our own hardware, the findings in the report should be applicable across various platforms and be valid for HW5 as well.

We've done tests with HW5 in parallel and noticed that when using an external supply at 5.3V all unlocks fail for us. Yet it works fine with the testing we've done under 5.0V.

petrutoader, 2023-12-14:
Attached you can find the report I've made regarding the different voltage levels and how it seems to influence unlocking and data r/w success. Please take note that the ROM integrity checks might fail due to improper timings in the read/write waveforms.

In addition to this, I have a Super Retro Trio 3+ that I've taken apart and it uses the superCIC project. When probing the voltage across the cartridge it reads a clear 4.7V which is in line with my report.

[Voltage Report.pdf](https://github.com/sanni/cartreader/files/13671623/Voltage.Report.pdf)

superbonaci, 2023-12-17:
@petrutoader it's mentioned here: [S-SMP](https://snes.nesdev.org/wiki/S-SMP), the game sound may vary depending on console temperature, not sure about cartridge' s one (may affect or not). The article says "may" but some experts say it's a fact: higher pitch with hither temperature.

Ancyker, 2024-04-27:
Does anyone have an SA1-based cartridge that's finicky to unlock that I can buy/borrow? I can return it to you after testing if you are in the USA. If it's more than $20-30 I will only borrow it (in other words, I'm not interested in buying an NA Super Mario RPG, lol).

I've got 3 of them now but none of them have any issues with consistently unlocking. I was thinking if the reason the unlocking fails on the OSCR is the voltage not dropping enough we could use VSELECT to toggle the voltage to 3.3V for a period to help it drop sooner. That may or may not help. Obviously, that's only a solution for units with VSELECT installed but it's something.

But I'm doubtful if the voltage drop is really our issue. The OSCR's bulk capacitor is, electrically speaking, very far from the slot and PIC -- I believe around 8-10+ inches. Even with me adding a 100 uF bulk capacitor to the CIC module I can't get the SA1 carts I have to fail to unlock. The OSCR has poor voltage regulation response at the slots, the droop is pretty awful and modern silicon would probably malfunction if it worked at all. Luckily this old tech isn't as sensitive.

Now seeing as most people building an OSCR use the cheap clock generators from Aliexpress (I can get these for less than $2/unit shipped), if the clock signal generator's quality has anything to do with whether unlocking is successful or not then that might explain things a bit. That is to say, maybe the people with issues have a bad clock generator. Not so bad that nothing works but rather one that's bad enough that the SA1 cares.

Anyway, yeah, all of this combined with how rarely people run into carts that won't unlock makes me think that the voltage issue is not why we are seeing carts that won't unlock. If not the clock generator then maybe something else. But that's all just a theory, I can't test any of this without a cart that has issues unlocking. Though, if I'm right about it being a bad component (i.e. the clock gen), then if someone sends me a cart that they have issues unlocking I won't have any issues with it.

Ancyker, 2024-05-09:
Alright, so here's a weird one. I finally encountered a PIC module that won't correctly dump my copy of PGA Tour. About half the time it fails to unlock and when it does unlock the dump is invalid:
[image]

Now, I originally wrote this module off as just a bad module, and in a way it is, as it was 1 of 10 I made and the others worked fine.

However, today that changed. I made a new Clock Generator PCB that includes the RTC on it as well with the idea that anyone could use it on any version of the OSCR with a Clock Generator header. Since I was making it specifically for the OSCR I opted not to include the footprint/pads for the BNC connectors and moved the clock generator IC as close to the clock pins as possible. I also used a 4-layer PCB.

Today the assembled modules I ordered came and I was testing them and they worked fine. I got curious and popped in that CIC module that wasn't working (the one in the image above) and gave it a shot. I didn't expect it to work but I figured why not. I'm sure glad I did, because:
[image]

To my complete and utter amazement, it worked. Since I had to remove the RTC IC from the main PCB, at first I thought maybe I had just done something else (i.e. reseating the adapter, cart, etc, did something) and figured it was a coincidence, so I put my original clock gen (Aliexpress) back and disabled RTC in the firmware, and sure enough _it doesn't work_:
[image]

Wanting to go further I tried a real Adafruit Clock Generator, and that didn't work either:
[image]

At this point I was thinking it must have just been a coincidence, and since I only tried with my clock gen once I put another, but not the same, of my clock gens in and tried again. It worked...
[image]

So I tried yet another one of them...
[image]

So, yeah. At this point, I think it's safe to say the clock generator fairly conclusively plays a role in whether or not SA1 games unlock. While I still think there is something wrong with this module as it doesn't consistently unlock the cart even with my clock gen (though when it does unlock the dumped ROM seems to always pass the checksums), the fact that changing it out lets it work is quite interesting. It seems fairly consistent as well, maybe more can come of this when I have time to mess with it and experiment more.

sakman55, 2024-10-06:
I'm in the process of finally building a HW5 unit.  My temperamental PGA Tour 96 (US) cart does not like HW5.  I'm using a 12F629 on a small PCB with an original Adafruit Clock Generator module.

I'll try to do some comparisons with my HW1, HW2, and portable SNES reader to see what might be the difference.

sakman55, 2024-11-30:
Update:  My temperamental PGA Tour 96 (US) cart dumps fine now on HW5.  Fully assembled HW5 with all the options in a Kytor shell.  My HW5 uses a small snesCIC 12F629 board.  After final assembly, I redid the clock calibration and found that the finicky cart dumps without a problem.