sanni/cartreader #735: screen does not display correctlly when in 3v mode

machineh3d, 2023-03-11:
this is my first project like this and i thought everything was going fine i was able to dump some GB carts and SNES carts 

but the moment i try and turn it on with the switch set to 3v the screen either looks glitchy or nothing at all 

5V
[image]

3v
[image]

closed 2023-03-13 by machineh3d

djbiohazard, 2023-03-26:
Mine is being weird too, It's fine at 5V setting, but once I switch it to 3V setting, it'll boot, briefly show a message, before showing a green solid screen.

machineh3d, 2023-03-27:
> Mine is being weird too, It's fine at 5V setting, but once I switch it to 3V setting, it'll boot, briefly show a message, before showing a green solid screen.

I fixed mine by replacing the MEGA 2560 pro with a new one. The one that wasn't working was a ATMEGA2560-16AU and I swapped it with ATMEGA256016U2 and now it works with 5v and 3.3v no problems at all

Ancyker, 2023-03-27:
You might be shorting 5V to VCC. The 3.3V comes from the LCD.

djbiohazard, 2023-03-27:
@Ancyker Problem is solved. 

My ADD brain caused me to desolder the diode instead of the fuse. 
Ofc the diode was long gone. Replaced it with 1N4148, and it's now working fine.  Did (eventually) remove the fuse.