sanni/cartreader #745: Dumped NES Games wont play in emulator.

FlyingMaster97, 2023-03-24:
I tried playing the NES games on my emulator, and it doesn't work. Yes, I did add the emulation header and cleaned the cartridges/72-pin connector port. Yet it just results in a grey screen in all games I've tested. I've tried a ROM from the internet, and that worked fine, so its not a problem with the emulator. Do you guys know of a solution?
[image]

sanni, 2023-03-24:
https://github.com/sanni/cartreader/wiki/Troubleshooting#when-is-a-rom-bad

nsx0r, 2023-03-24:
Which game is it and what CRC32 did your dump get?

FlyingMaster97, 2023-03-24:
Thanks! There was a bad solder. Got it off and worked perfectly!