sanni/cartreader #749: Wiki says N64 slot should be 2.5mm pitch, PCB revision states 2.54mm?

chunkysteveo, 2023-03-24:
Wiki What To Buy page says N64 slot should be 2.5mm pitch, warns about sellers sending 2.54mm pitch, links to a 2.5mm pitch item. 

Revision on 6 slot board 2 months ago states N64 slot changed from 2.5 to 2.54mm?

Which pitch do we need? 

Is the individual N64 slot board same pitch as the 6 slot board?

Thanks, Steve

Ancyker, 2023-03-24:
> Revision on 6 slot board 2 months ago states N64 slot changed from 2.5 to 2.54mm?

[image]

chunkysteveo, 2023-03-26:
Ahhhh..!!! Hahaha! My mistake, looking without reading lol! Thanks for the feedback, I'll continue with my part purchase. Keep up the great work!!

closed 2023-03-26 by chunkysteveo