sanni/cartreader #777: Flashrom Progammer : Can't flash + Menu error 255

Slyazku, 2023-04-18:
When using the flashrom PCB adapter with a 29L3211 inside, plugged into the SNES port, I've got no result.
Blanckcheck, reading, etc takes 0sec and dumped files are blank.
When the first operation is complete, and after "Press Button..." prompt, I've got a 
"MenuError 
Mode = 255
Press BUtton..."
Then to the tiel screen.

Did someone managed to flash a flashchip with 12.4 HW5 ?

Thanks !

sanni, 2023-04-20:
There was an error in the menu resulting in the flash ID function being skipped. Fixed now.

Also since there are currently no HW4/5 flash adapters you have to modify the HW3 adapter:
- cut the trace going to SNES CPU Clock and rewire Flashrom OE to SNES CS
- cut the trace going to SNES IRQ and rewire Flashrom WE to SNES WR

(you need to scratch away the solder mask of the edge connector pins to expose the copper to solder to)

[image]
[image]
[image]

sanni, 2023-04-20:
You can update to the current github code like explained here: https://github.com/sanni/cartreader/wiki/Troubleshooting#updating-to-the-latest-github-code

Slyazku, 2023-04-21:
Thanks !
I may be late but did you plan to make the eeprom  section appear again in the wiki, since the menu is able to do it ?

closed 2023-04-21 by sanni

sanni, 2023-04-22:
You can always enable the eprom menu option by setting #define enable_FLASH16 in [config.h line 108](https://github.com/sanni/cartreader/blob/master/Cart_Reader/Config.h#L108)

I have updated the [M27C322 wiki entry](https://github.com/sanni/cartreader/wiki/M27C322-Adapter) with the notion that it doesn't work, just so people don't order it and get disappointed.