sanni/cartreader #781: Mario RPG SA-1 HiRom issue

Slyazku, 2023-04-27:
Well I have this bad identification of Super Mario RPG, it should be Lowrom. Dump works but CRC fails and I can't read rom file in an emulator.
I use HWrev 5 and tried 2 diferent boards of SNES Cic and 2 different clock generators, one working perfectly on HW rev3
[image]

sanni, 2023-05-04:
The OSCR dumps SA1 as if was HiROM.

That being said, SA1 code is still far from perfect. One thing you can try is to lower the clocks.
A real SNES seems to run at only  5.3MHz for EXT and 2.6MHz for CPU clock when reading SA1:

[image]

Change [line 484](https://github.com/sanni/cartreader/blob/master/Cart_Reader/SNES.ino#L484)

````
 if (i2c_found) {
    //Set clocks to standard or else SA-1 sram writing will fail
    clockgen.set_freq(2147727200ULL, SI5351_CLK0);
    clockgen.set_freq(357954500ULL, SI5351_CLK1);
    clockgen.set_freq(307200000ULL, SI5351_CLK2);
  }
````
to:

````
  if (i2c_found) {
    //Set clocks to standard or else SA-1 sram writing will fail
    clockgen.set_freq(536931800ULL, SI5351_CLK0);
    clockgen.set_freq(268465900ULL, SI5351_CLK1);
    clockgen.set_freq(307200000ULL, SI5351_CLK2);
  } 
````

Maybe it helps.

Also make sure you got Rev3 of the Six Slot Adapter or apply the [fix for Rev1/Rev2](https://github.com/sanni/cartreader/issues/481).

Slyazku, 2023-06-25:
Sorry for the late answer. Just wanted to let know the fix works perfectly, I can dump the Rom despite the CRC32 Error - Error Database Missing.
Thanks !

sanni, 2023-06-26:
With fix you mean the hardware fix for Rev1/Rev2 or do you mean changing the code like shown above?

closed 2023-06-26 by sanni