sanni/cartreader #785: [NES] SRAM is only successfully dumped after having dumped the ROM first

rsvensson, 2023-05-03:
If I try to dump the SRAM of an NES/FC game without first having dumped the ROM, it won't actually dump the SRAM. Choosing the `Read SRAM` option without dumping the ROM first will instantly show only `Press Button...` on the screen and no save file exist on the cartridge. However if I first dump the ROM and then chose `Read SRAM` afterwards it does show that the SRAM was dumped properly on the screen, and the file exists on the SD card.

So far I've only tested this with SNES games as well, and there it does dump the SRAM correctly without first having to dump the ROM.

Here's some relevant lines from my log where I first try to dump the SRAM without dumping the ROM, followed by dumping the SRAM after dumping the ROM.
```
OSCR HW5 V12.4

[+] NES/Famicom

Searching database
for 470814D0 or B3CD71E2...

NES CART READER

CURRENT SETTINGS

MAPPER:   1
PRG SIZE: 128K
CHR SIZE: 128K
RAM SIZE: 8K

Press Button...

[+] Read Sram

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/ZeldaIITheAdven/7/...
[*******************]
CRC32... 47B6A39F -> Zelda II - The Adventure of Link (Europe) (Rev 2).nes

Press Button...

[+] Read Sram

RAM FILE DUMPED!

CRC: B8B04237

Press Button...
```

sanni, 2023-05-04:
Can you try if it works with [V10.2](https://github.com/sanni/cartreader/releases/download/V10.2/V10.2_Portable.zip)?

sanni, 2023-05-04:
As a band-aid you also try replacing the CreateRAMFileInSD function in V12.4 with the one from V10.2:

[In NES.ino around line 1176](https://github.com/sanni/cartreader/blob/master/Cart_Reader/NES.ino#L1176)

Replace:
````
void CreateRAMFileInSD() {
  myFile = createNewFile("RAM", "bin");
}
````

with:

````
void CreateRAMFileInSD() {
  char fileCount[3];
  strcpy(fileName, "RAM");
  strcat(fileName, ".bin");
  for (byte i = 0; i < 100; i++) {
    if (!sd.exists(fileName)) {
      myFile = sd.open(fileName, O_RDWR | O_CREAT);
      break;
    }
    sprintf(fileCount, "%02d", i);
    strcpy(fileName, "RAM.");
    strcat(fileName, fileCount);
    strcat(fileName, ".bin");
  }
  if (!myFile) {
    LED_RED_ON;

    display_Clear();
    println_Msg(F("RAM FILE FAILED!"));
    display_Update();
    //print_Error(F("SD Error"), true);

    LED_RED_OFF;
  }
}
````

For the real fix we have to figure out why the function createNewFile is crashing.
````
FsFile createNewFile(const char* prefix, const char* extension) {
  char filename[FILENAME_LENGTH];
  snprintf_P(filename, sizeof(filename), _file_name_no_number_fmt, prefix, extension);
  for (byte i = 0; i < 100; i++) {
    if (!sd.exists(filename)) {
      return sd.open(fileName, O_RDWR | O_CREAT);
    }
    snprintf_P(filename, sizeof(filename), _file_name_with_number_fmt, prefix, i, extension);
  }
  // Could not find an available name, recompose the original name and error out.
  snprintf_P(filename, sizeof(filename), _file_name_no_number_fmt, prefix, extension);

  LED_RED_ON;

  display_Clear();
  print_Msg(filename);
  println_Msg(F(": no available name"));
  display_Update();
  print_FatalError(sd_error_STR);

  LED_RED_OFF;
}
````

If it says "Press Button" without dumping anything or if it hangs on "RAM FILE DUMPED" while creating a wrongly named "______~1.___" file seems to be dependent on whether a RAM.bin is already present or not in the NES/ROM folder.

rsvensson, 2023-05-04:
So I flashed both of the readers with V10.2, and now they both consistently dump the RAM correctly without first having to dump the ROM. There's a bit of a delay after selecting `Read SRAM` as if it is initializing something, then it succeeds.

> If it says "Press Button" without dumping anything or if it hangs on "RAM FILE DUMPED" while creating a wrongly named "______~ 1.___" file seems to be dependent on whether a RAM.bin is already present or not in the NES/ROM folder.

Hmm on V12.4 it does not create RAM.bin files but rather .txt files that are named the same as the game's ROM folder. I thought this was odd but I just renamed them with a .sav extension to use in emulators and it worked. There was never any RAM.bin file on the SD card originally since V12.4 was the version installed initially on these readers. I can also not see any `_____~1.___` files although maybe it crashed before it started the write. I was using the same SD card on both readers btw.

Since this is only a minor inconvenience (usually I'd dump the SRAM after dumping the ROM anyway, I just forgot with one game) I think I'll hold out for the proper fix for now. Thank you for your time and suggestions. :)

rsvensson, 2023-05-04:
Oddly enough after I put V12.4 back it just stopped writing NES saves to the card entirely even though it seemed to succeed judging by the messages. I had left a RAM.bin file on there but I removed it afterwards and tried again. It still wouldn't write the save though. After reformatting the SD card it is now writing them out as .txt files again like before. Just thought I'd let you know.

sanni, 2023-06-26:
I have added the temporary fix to the current github code. Please [update](https://github.com/sanni/cartreader/wiki/Troubleshooting#updating-to-the-latest-github-code) and try again.

closed 2023-06-26 by sanni