sanni/cartreader #799: Derby Stallion 96 Satellaview Enabled (SFC) problem dumping on HW5 V12.4

Prominos, 2023-05-28:
It seems like there might be a software bug dumping Derby Stallion 96 for super famicom on HW5 V12.4

[image]
[image]

This is a 24M (3MB) chip. When Dumping with HW5 V12.4 the first 2MB are dumped correctly but the last 1MB (starting from 0x200000) returns all FF's. 

I have tried on 2 different OSCR HW5 V12.4 with the same result 2MB ok last 1MB all FF's

I have tried on OSCR HW3 V5.5 it correctly dumped the entire cart on the first try.

I have tried downgrading one of my HW5 to V8.4 and dumping again and this time it correctly dumped on HW5 too on the first try.

splash5, 2023-05-28:
Please check #760, sanni should have updated the code. Update SNES.ino from repo and see if it helps.

Prominos, 2023-05-28:
Oh thanks, I should have double check that there was already an issue about that before opening a new one. My apologies!

closed 2023-06-26 by sanni