sanni/cartreader #801: Atari - Issue dumping Pitfall II

PsychoFox11, 2023-06-01:
I don't feel like there's much traction on Atari but I'm trying to go there haha, and could use help if any is out there.

Pitfall 2, as far as I understand, was the only original retail game with 3 banks to switch between, with 2 4k banks and a 2k bank. When selecting it (Atari requires you to manually select the game each time), it chooses 10k as the size. However, I get a consistent, but different result compared to the no-intro verified dump. the ROM is about 99% the same with a couple small differences, however, and I think this may be on the Sanni, the dump listed on No-intro is 10,495 bytes, while what I get is 10,240 bytes. I believe the initial selection dictates the size, so is the Sanni underdumping?

Also on a similar note, my dump of Realsports Baseball (8k) is consistent, but a single byte different than the no-intro dump. Oddly it is the 8th to last byte before the end of data and empty space starts. Is this a bad dump or possibly a different version? Seems odd to have an exact dump aside from one byte consistently, and the location of the byte is weird. Maybe bad cart also? Size of both is the same.

Worth noting both of these sanni dumps work in emulators fine, but I guess there's a chance I just haven't hit the part of the code that would be affected by these differences.

sanni, 2023-06-26:
You might have a better chance asking skaman in [the Arduino forum thread](https://forum.arduino.cc/t/rom-reader-for-super-nintendo-super-famicom-game-cartridges/155260/1615) since those issues need a deeper understanding of the Atari code and I'm not sure if someone else here has worked with Atari yet.

PsychoFox11, 2023-06-28:
I reached out to skaman and pointed them here. I'm talking with them and providing some info to help look into it. Definitely seems to be an issue though, feel it would be useful to keep it here for others to be aware of until it is resolved.

sakman55, 2023-07-05:
I can confirm that there currently is an issue with the F6 code.  The bank switch register area needs some additional code like what was done for the F8 carts.  I have a working code fix but I'm waiting to test an F4 cart to see if that mapper also needs a similar fix.

I cannot duplicate the reported errors with F8 carts Ms. Pac-Man and RealSports Baseball.  Both test carts that I picked up dump properly and match the known dumps.

Dumps of Pitfall II will not match the known dump.  The register area at 0x1000 are random on boot.  I chose to output 0xFFs in this area as each read will produce a different output.  The "missing" 255 bytes tacked on the end of the known dump is some RNG data generated by the DPC chip.  The RNG aka "poly" bytes do not exist in the ROM so are not included in the dump.

PsychoFox11, 2023-07-05:
I included this in a message to Skaman, but just to note my findings here (and if anyone else wants to share theirs). I have 2 copies of RealSports Baseball, and two OSCR HW5 dumpers (vselect being the only difference), running the latest 12.6.
They always get this error, in the same place, on either cart, in either OSCR. I also have a Maxflash dumper from AtariMax and RealSports Baseball as well as Ms. Pac-Man dump fine from it. Only the OSCR gives me the issue with these 3 carts.

jaffa225man, 2023-10-08:
My Ms. Pacman dumps just fine as skaman suggests, and likely my Pitfall II does too.

I'm getting an md5sum of Ms. Pacman as this:
87e79cd41ce136fd4f72cc6e2c161bee
And my (10240 byte) Pitfall II is gives this as its md5sum:
37ea16edda0bc46f5527c3890ea121ca

closed 2024-04-29 by PsychoFox11