sanni/cartreader #805: Cannot scroll and 3v does not work

TH3GOD88, 2023-06-04:
I built the HW5 and installed everything correctly. The screen powers on and shows all of the systems even though I marked out the ones I would not be using. When I try to power it on with 3v, it flashes red and does nothing. I cannot seem to scroll at all in the menu.

Ancyker, 2023-06-04:
Sounds like you are shorting 3V to ground somewhere. Check with a multimeter.

TH3GOD88, 2023-06-04:
Okay, it seems to be the connecter between the boards. I still cannot scroll through the menu though.

PsychoFox11, 2023-06-04:
Assuming mine is correct it will always show all the systems. I didn't really understand why I need to enable them as they are already listed and trying to pick a disabled one just gets a message to enable it. It would be useful if it only listed enabled ones though. You should be able to scroll though...

TH3GOD88, 2023-06-05:
I twist the nob and doesn't move. I can push it in, but GBA is first, and I need N64.

TH3GOD88, 2023-06-05:
Okay, I figured it out. The ribbon cable did not make a good connection and now I can scroll. Thank you both for the help.

closed 2023-06-05 by TH3GOD88