sanni/cartreader #807: [NES] issue reading writing NES saves

BrendonMelling, 2023-06-06:
Hello I am having issues reading NES saves. I have several battery backed up carts such as dragon warrior 1-4 and final fantasy. When I insert the cart it identifies the cart correctly. It reads the ROM but cannot save files onto the device. I am using a V3 reader (pre built by save the hero builders) SNES and Gameboy works fine, just NES I’m having issues with. Thank you for your assistance.

PsychoFox11, 2023-06-07:
Hi, first is there a reason you haven't moved to 12.4? Not sure if it would make a difference, but I've never used 12.3. I got mine right when 12.4 went out of beta. I've been using the 12.5 files which aren't out as a stable release, but I can share my experience with 12.4 in any case.
First, it won't make a .sav file and it won't be in the game directory. It places the save in the main NES folder, and gives it a .txt extension, so you'll need to rename it. For example you could get FinalFantasy.txt.

Also, there is a known issue (at least as of 12.4, which I believe exists in 12.3 as well) where you have to dump the ROM, then the save, or it won't work. If you try the save without dumping the rom first, it just doesn't work.
If it's none of this, then I'm not sure!

Prominos, 2023-06-09:
Link to known issue: #785

sanni, 2023-06-26:
I have added the temporary fix to the current github code. Please [update](https://github.com/sanni/cartreader/wiki/Troubleshooting#updating-to-the-latest-github-code) and try again.