sanni/cartreader #812: Add Bypass Capacitors for Voltage Smoothing

RWeick, 2023-06-14:
I’ve recently cloned the N64 GameShark and released it publicly. It uses an Altera EPM240, and it doesn’t work with the sanni cart reader until I program it with verilog to sample the input signals more times than is compatible with the speed at which the signals are executed on an N64. This has caused the realization that the voltage on the sanni cart reader is not stabilized, and for better compatibility and stability, I am recommending the design be updated with bypass capacitor to smooth the voltage running to attached devices. This should allow the N64 GameShark clone to function with the verilog meant for the N64 as well.

Ancyker, 2023-06-28:
Have you tried adding a bulk capacitor to the CIC board?

[image]

Ancyker, 2023-08-08:
Ordered for testing:

[image]

Ancyker, 2024-03-22:
I figured it's been a while so I should post an update on this as I have made some progress. Here are the things I've found out and my progress thus far:
* The six-slot adapter SMT variant has been replaced by the seven-slot adapter.
* Adding the decoupling capacitors smoothed out the voltage droop on VCC but did effectively nothing for the signal integrity. As everything I tried failed to change this I can only assume that it's due to how far the slots are from the Arduino.
* Since I gave each slot a decoupling capacitor it made the bulk capacitor unnecessary (no measurable difference in voltage droop on VCC or signal integrity with or without a 100uF polymer capacitor). Because of that, the seven-slot adapter does not have a bulk capacitor. If you are wondering, with only the bulk cap the voltage droop did improve but not as well as it does with the decoupling capacitors. It's effectively non-existent with the decoupling capacitors (so small it was hard to reliably capture with my scope). While with the bulk capacitor, it was still quite noticeable, albeit smaller. If desired, placing a bulk capacitor on the CIC board for the six-slot adapter, as I do on my builds, also produces this result.
* The seven-slot adapter *only* has an FPC board-to-board connection and does not support using pin headers. I tried to route it with both but neither I nor the auto-router could do it even when using all 4 layers for signal traces (primarily due to the traces having to fit between the headers' pins when fanning out) -- even routing this board without the pin headers was a bit of a pain. So, it would need to be a 6-layer PCB to offer both the FPC and pin-header connections, which is so expensive you might as well buy a preassembled 4-layer seven-slot.
  
  Getting 4-layer boards from JLCPCB would be $7 for 5, while 6-layer PCBs are $60 for 5 ($86 for 8-layer). Since that's about what it costs to pay JLCPCB for assembly of the 4-layer version it makes no sense to make a 6-layer version at this time. If prices for 6-layer PCBs come down I'll revisit making one.
* The new revision of the main PCB makes it a 4-layer PCB. This might help with the signal integrity issues but the first batch of the new main PCB had an error on it and didn't work. I will reorder within the next few days.
  
  It also offers the ability to put the ATmega2560 and clock generator directly onto it, so this might help clean up the signals even further. This is somewhat difficult for your average DIYer, but don't worry if this sounds a bit much for you as it still supports using the modules just like the current revision does.
  * Note: The six-slot adapter remains a 2-layer PCB and cannot make use of the FPC cables. If ordering from JLCPCB and you add it to your cart first it will cost $2. Then the old (current, Rev 5) main PCB would cost $4 to add but the new 4-layer version will cost $7. That means this change only increases the overall cost by $3. If this version of the board cleans up the signals I think this is worth it.
    
    However, if it does not, then I will consider making it a new variant rather than replacing the current main PCB with it as even though $3 isn't a ton it's adding extra cost for no benefit if not putting things directly onto the board or making use the FPC connections.
* And finally, if this doesn't clean up the signals enough but they do show some improvement I'm going to start playing around with an SMT-only main PCB that does away with the ability to use the now integrated modules (the ATmega and clock generator, specifically). I might also try a single-board OSCR (six-/seven-slot and main PCBs combined into one PCB).
  
  The main downside of a single-board OSCR is that it would be larger than 100mm x 100mm which will significantly increase the cost of the PCB and make it incompatible with most existing cases/enclosures/frames.
  
  The cost of a 125mm x 100 mm 2-layer PCB isn't bad, about $9 for 5, but if it needs to be a 4-layer PCB it would cost about $30 for 5. At scale the cost drops a bit, if ordering 100 PCBs 2-layer PCB would cost $70 for 100x100mm or $83 for 125x100mm. 4-layer PCBs see a similar bump costing $104 for 100x100mm and $124 for 125x100mm.
  
  Sellers of PCB kits should have no issues offering a single PCB for the same (or lower) cost as a set containing a main PCB + six-slot PCB + CIC adapter PCB. So, for anyone who wants a single-board version, the cost shouldn't be an issue.

As always, questions/comments are welcome. :3

zombie343, 2024-06-28:
Following, and definitely interested in increased electrical compatibility