sanni/cartreader #818: vSelect not working beyond 12.4

PsychoFox11, 2023-06-25:
I know 12.4 is the latest stable release, but wanted to note this as I've been using 12.5 and now 12.6. I just got a new Sanni with vSelect in addition to the one I already had (both HW5). I always comment out the line to enable vSelect in the appropriate file (different place in 12.4 than later). I'm enabling that, clockgen, and 4 additional systems for adapters.

Using a multimeter, the voltage only ever switches to 5v when using 12.4. Anything later, 12.5 or 12.6, and it stays at 3.3. I've tested this on the Coleco and Intellivision adapters while dumping a real cart. I've tested this on the SNES and Genesis slots while dumping with an empty slot (easier to get to with the multimeter). It only goes to 5v on 12.4.

It isn't especially apparent as many of the built-in slots that are 5v still usually give decent dumps at 3.3v. I have tried SNES and NES and they dumped just fine. I also tried them in my older one with the physical switch set to 3.3v and they dumped fine there as well. Probably not the best for them, but they don't make the issue obvious.

However, Intellivision and ColecoVision will not give me consistent CRC32 at all, unless I use 12.4, or my older HW5 with the manual voltage switch. They seem to really need 5v. Just trying all my adapters, Game Gear, PCE, and Atari gave me good dumps (only tried one cart each) at 3v as well. So even if you're getting good dumps, and have vSelect, you should probably stick with 12.4 until this is changed, for any 5v systems. At least test with a multimeter, as I doubt it is any issue with my setup, given that 12.4 works. I think it would affect everyone.

To note how I'm flashing, I'm downloading the file at:
https://github.com/sanni/cartreader/archive/refs/heads/master.zip

I'm replacing the contents of my portable/sketchbook/Cart_Reader folder with the contents of the above file's Cart_Reader folder's contents, then launching the Arduino software, setting my config options, and flashing.

By the way I've checked and reflashed back and forth a lot and am positive I'm uncommenting the like to enable vSelect every time, and recommenting it for my older reader. I've verified the heck out of that, unless more than one line needs uncommented in 12.5/6 that I'm missing.

sanni, 2023-06-26:
Can you try the latest commit?  I moved the automatic voltage switching forward in the code so it now switches in the setup function instead of the menu function of the module.

closed 2023-06-26 by Ancyker

Ancyker, 2023-06-26:
This is now fixed. I made the image below to explain what the issue was, for anyone curious:

[image]

I probably shouldn't have used "volts" as the input, I should change that to something like "requestedVoltage" or "newVoltage" as that would have likely made the issue more clear. My fault, this is why what you name variables is important and I definitely know better 😅

commit 9fb7af3f (2023-06-26): Fix automatic voltage selection (fixes #818)

Fix for automatic voltage selection only working when 3V3FIX was enabled.