sanni/cartreader #826: Famicom: Unable to dump save file from Mother

JoshuaRamey, 2023-07-13:
As the title says, I seem to be unable to dump my save file from this game. If I try dumping the save then the only text that displays is "push button", but if I dump the rom first then it looks like it dumps the save however it isn't there when I check the SD card.
[image]

sanni, 2023-07-13:
Can you try [updating](https://github.com/sanni/cartreader/wiki/Troubleshooting#updating-to-the-latest-github-code) to the current development code and try again, there was a bug in V12.4. The save should be called ram##.bin

JoshuaRamey, 2023-07-14:
Thanks. I'm away from home for the next few days but I'll give it a try when I get back.

JoshuaRamey, 2023-07-20:
Thank you for the help. I was able to dump my save using the latest release.

[image]

closed 2023-07-20 by sanni