sanni/cartreader #829: Unable to dump Tengen Tetris for the NES

retrovik, 2023-07-18:
It finds it in the database but only creates a 1kb file after reading the cart. I have verified I can dump this cart with another device and it gives the correct crc and size.

----------------------------------------------------------
OSCR HW3 V12.0

[+] Add-ons

[+] 80s Consoles

[+] NES/Famicom

Searching database
for F4CDC28B or 3617FCC0...

NES CART READER

CURRENT SETTINGS

MAPPER:   148
PRG SIZE: 32K
CHR SIZE: 16K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/Tetris/233/...
[*******************]
CRC32... 00000000 -> Not found

Press Button...

retrovik, 2023-07-18:
Update:

If I manually set the mapper and PRG/CHR it will product a correct dump.

-----------------------

OSCR HW3 V12.0

[+] Add-ons

[+] 80s Consoles

[+] NES/Famicom

Searching database
for F4CDC28B or 3617FCC0...

NES CART READER

CURRENT SETTINGS

MAPPER:   148
PRG SIZE: 32K
CHR SIZE: 16K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/Tetris/235/...
[*******************]
CRC32... 00000000 -> Not found

Press Button...

[+] Change Mapper
Set Mapper manually
Set PRG Size
Set CHR Size
Set RAM Size

NES CART READER

CURRENT SETTINGS

MAPPER:   3
PRG SIZE: 32K
CHR SIZE: 16K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/CART/236/...
[*******************]
CRC32... 343C7BB0 -> Tetris (USA) (Tengen) (Unl).nes

Press Button...

Ancyker, 2023-07-18:
Could you take a pic of the PCB by chance?

retrovik, 2023-07-18:
Unfortunately, the center back screw is under the label, and I do not want to damage this cart (because of the rarity). Otherwise, I would be happy to open it.

Ancyker, 2023-07-18:
Fair enough. It does appear 3 should be the correct mapper anyway.

@nsx0r yeah?

nsx0r, 2023-07-18:
mapper 148 script has been fixed

closed 2023-07-18 by nsx0r