sanni/cartreader #836: Blank screen on one line only

BenoitCaron, 2023-08-05:
Hi, this is the second HW5 I'm assembling. The first one works fine, but this one doesn't.

Firmware 12.8 beta (red display)
- when started without an SD card, the screen displays "SD Error" as expected, but the the text slowly fades away (not shown on video)
-  when started with an SD card (correctly configured), the screen briefly shows a row of pixels, then nothing. Once did I manage to have a single line of text showing ("Game boy"), but even then the bottom of the text was truncated.

Firmware 12.4 (green display):
- when started without an SD card, the screen goes full green
- with the SD card, same as above

Is my screen defective? I'm pretty sure I followed instructions to the letter, + my soldering doesn't seem bad :/

https://github.com/sanni/cartreader/assets/5237965/57268ce8-2fe7-4593-bfcc-f0d70a47cf39

https://github.com/sanni/cartreader/assets/5237965/da9552e2-79ea-4fc8-bbaf-a481c8cb4ed6

https://github.com/sanni/cartreader/assets/5237965/561369b6-04ea-447c-ab1f-cc5f5ac9758c

https://github.com/sanni/cartreader/assets/5237965/66e48d40-732a-40e6-b33d-fb2a290844a0

sanni, 2023-08-05:
Looks like a short circuit somewhere near the SD slot. MOSI and SCK seem to be the only pins shared between the LCD and the SD card according to the [schematics of the lcd module](
https://github.com/makerbase-mks/MKS-MINI12864-V3/blob/main/hardware/MKS%20MINI12864%20V3.0_002/MKS%20MINI12864%20V3.0_002%20SCH.pdf
), so I would start there.

BenoitCaron, 2023-08-05:
So these two points? They're not shorted :(
[image]

Ancyker, 2023-08-07:
Are these pinn actually soldered?

[image]

BenoitCaron, 2023-08-07:
Yes, they are.

BenoitCaron, 2023-08-09:
Well my return window was closing so I returned the display and ordered a new one. I'll close this issue for now and will update it when the new display arrives.

closed 2023-08-09 by BenoitCaron

BenoitCaron, 2023-08-21:
Follow-up: received my new screen and it now works as expected, no changes whatsoever to the main board. First screen was defective. Thank you all for your help!