sanni/cartreader #841: CRC32 not found for MD/Genesis NHL 98

LucidusMC, 2023-08-23:
### OSCR firmware version

12.4

### OSCR hardware version

HW3

### Attach OSCR_LOG.txt file here

[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/12423831/OSCR_LOG.txt)
[image]
[image]
[image]

### System used

Sega Mega Drive/Genesis

### Describe the issue.

Trying to dump MD/Genesis NHL 98 on HW3 with 12.4. (See attached) and getting not found and I can't seem to run the bin file as it is written. I have also attached pictures of the PCB. I have successfully played the game on my console (I know this doesn't always mean the cartridge is stable) and cleaned the contacts thoroughly.

OSCR HW3 V12.4

[+] About

Cartridge Reader
github.com/sanni
2023 Version 12.4

Press Button...

OSCR HW3 V12.4

[+] Mega Drive/Genesis

[+] Game Cartridge

Cart Info
 
Name: NHL98
Size: 4 MBit
ChkS: 5B3A
Sram: 256 KBit
 
Press Button...

[+] Read Rom

Saving to MD/ROM/NHL98/31/...
[*******************]
Internal checksum...OK
CRC32... 5CB4A9AD -> Not found
Press Button...

Ancyker, 2023-08-24:
Is the CRC the same every time you dump it?

LucidusMC, 2023-08-24:
> Is the CRC the same every time you dump it?

Yes

Ancyker, 2023-08-24:
Are you in the Discord?

LucidusMC, 2023-08-24:
Yes. Just joined.

commit ec3d0481 (2023-08-24): Update MD.ino

Add "NHL 98 (USA)"
Fixes #841
closed 2023-08-24 by Ancyker