sanni/cartreader #869: Does not compile with HW5 and Emerson Arcadia enabled

slewantoski, 2023-11-20:
### OSCR firmware version

13.0

### OSCR hardware version

HW5 Rev5

### System used

Emerson Arcadia 2001

### Describe the issue.

I've defined HW5 and enable_ARC in the config.h an then 13.0 does no longer compile:

Errors:
`HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_25'
sketch\ClockedSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_26'
sketch\ClockedSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `Serial0_available()'
sketch\ClockedSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Arduino Mega or Mega 2560.
`

If i disable Emerson Arcadia Support it just compiles fine.

slewantoski, 2023-11-21:
With this change it compiles again. Thanks.

sanni, 2023-11-21:
Found the issue and now it should also compile with #define ENABLE_UPDATER 
Updated the release to 13.1: https://github.com/sanni/cartreader/releases/tag/V13.1

closed 2023-11-21 by sanni