sanni/cartreader #870: Vselect doesn't work as expected

Snkfear, 2023-11-22:
### OSCR firmware version

13.1

### OSCR hardware version

HW5 Rev6

### System used

all/none/doesn't matter

### Describe the issue.

I've populated everything like the wiki says but when I do the last test, Pin 8 doesn't switch to 3.3V it remains 5V.
When I poer on the system, I measure the expected voltages:
Pin 6 and GND -> 5V
Pin 8 and GND -> 3.3V
Pin 7 and GND -> 3.3V
Once I select "Mega Drive/Genesis" :

Pin 6 and GND -> 5V
Pin 8 and GND -> 3.3V (should be 5V)
Pin 7 and GND -> 3.3V

All smd components are soldered as they should and there isn't any bridge between the legs of the IC.

What else should I check?

sanni, 2023-11-23:
I don't use Vselect myself so I can't really help but to me it sounds like a typo in the Wiki. 

Pin 8 is a voltage input so it should always be 3.3V. 
Pin 7 should be 3V in the main menu and 5V after Mega Drive/Genesis->Game Cartridge was chosen.

Can you check the voltage on pin 7 again after Mega Drive/Genesis->Game Cartridge was chosen?

Snkfear, 2023-11-23:
Yeah, like you said, it was a typo on the Wiki. It's working great and I've dumped GBA/GBC game without issues.

The only thing that annoys me is that the menu isn't displayed correctly.

[image]
[image]
[image]
[image]
[image]

But as you can see on the last photo, some menus are displayed perfectly.

sanni, 2023-11-25:
Is there a pattern like it only happens with 3V systems but 5V systems are ok?

The Vselect by default switches the Cart Reader to 3V and only after a 5V system has been setup it switches to 5V.
In case of Mega Drive, the "Mega Drive Reader" menu would be the first menu where the Cart Reader running at 5V.
By this logic the menu should be ok again e.g. when you select Game Boy (Color).

If this is the case then it might be that your Mega Pro can't run at 3V. This seems to happen more frequently in the last years.

Snkfear, 2023-11-25:
You nailed it!

[image]

This is weird because I ordered the mega with a row of 5 smds according to the Wiki... and I've dumped GBA and N64 games that work without issues.

Ancyker, 2023-11-25:
@Snkfear try enabling 3V3FIX in the config. It might help as I wrote it to try to address the 3.3V issue but no one has reported if it helps or not.

```c
/* [ 3.3V Stability Fix (3V3FIX) ---------------------------------- ]
    Enable this if you are having stability issues when using 3.3V,
    works best with VSELECT.

    If not using VSELECT, always turn the cart reader on with the
    voltage switch set to 5V and switch to 5V before selecting a
    cartridge from the menu.
*/

#define ENABLE_3V3FIX
```

Snkfear, 2023-11-26:
> @Snkfear try enabling 3V3FIX in the config. It might help as I wrote it to try to address the 3.3V issue but no one has reported if it helps or not.
>
> ```c
> /* [ 3.3V Stability Fix (3V3FIX) [...]

The fix solved my problem!

[image]

I had to remove the mega and bridge the fuse because the board was restarting every time the ide started uploading the sketch... I guess it's because of the Vselect.

By the way, would  be great to add test points for the TPS in future hardware revisions, it was a nightmare to test each leg for shorts.

closed 2023-11-26 by Snkfear