sanni/cartreader #896: N64 Reflashing a N64 Repro  Clock Generator not found

ptromig, 2024-01-28:
### OSCR firmware version

DIY Sanni Cart Reader V5 Retro Game Card Dumper Open Source Cartridge Reader Whole Kits

### OSCR hardware version

HW5 Rev6

### Attach OSCR_LOG.txt file here

OSCR HW5 V13.1

[+] Self Test (3V)

[+] About

Cartridge Reader
github.com/sanni
2023 Version 13.1

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Clock Generator not found

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Clock Generator not found

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

[+] About

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Clock Generator not found

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

Self Test

Remove all Cartridges
before continuing!!!

Press Button...

Testing 1K resistor 
Testing short to GND
Testing short between pins
Testing Clock Generator

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Clock Generator not found

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...
n64.txt missing

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

[+] About

Cartridge Reader
github.com/sanni
2023 Version 13.1

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

Self Test

Remove all Cartridges
before continuing!!!

Press Button...

Testing 1K resistor 
Testing short to GND
Testing short between pins
Testing Clock Generator
Error: Clock Generator
not found

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: MARIOKABT&4
Serial: NKDJ
CRC1: 6BFF4748
Cartridge unknown

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Please reseat cartridge

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Please reseat cartridge

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Clock Generator not found

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: M@RIOKACT&4
Serial: NKDJ
CRC1: 6BEE4748
Cartridge unknown

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: M@RHOKACT'4
Serial: NKDJ
CRC1: 6BEE4748
Cartridge unknown

Press Button...

[+] 12 MB

[+] Read ROM

Saving to N64/ROM/GPERROR/68/...
[*******************]
CRC32... 6F104EB3 -> Not found
Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Gameshark

Check cart connection
Unknown Flash ID
Press Button...

[+] Nintendo 64 (3V)

[+] Flash Repro

Please reseat cartridge

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: M@RHOJAST74
Serial: NKTJ
CRC1: 6BFE4758
Cartridge unknown

Press Button...

[+] 12 MB

[+] Read ROM

Saving to N64/ROM/GPERROR/69/...
[*******************]
CRC32... E03E511F -> Not found
Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Please reseat cartridge

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Clock Generator not found

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Unknown flashrom
ID: 2672 1F40 1240
 
Press button for
manual config
This will erase your
Repro Cartridge.
Attention: Use 3.3V!

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Please reseat cartridge

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Please reseat cartridge

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Please reseat cartridge

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: 
Serial: 
CRC1: 00000000
Cartridge unknown

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: 
Serial: 
CRC1: 00000000
Cartridge unknown

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: 
Serial: 
CRC1: 00000000
Cartridge unknown

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

GAMEPAK ERROR

Title: S_A_H_B_OTHTR@ P
Serial: ALJ
CRC1: 67C007C0
Cartridge unknown

Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

Title: Blastdozer
Serial: NBCJ
Revision: 0
ROM Size: 8 MB
Save Type: 4K EEPROM
CRC1: 65234451
 
Press Button...

[+] Read ROM

Saving to N64/ROM/Blastdozer/70/...
[*******************]
CRC32... 081A3641 -> Blastdozer (Japan).z64
Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

Title: SMASH BROTHERS
Serial: NALJ
Revision: 0
ROM Size: 16 MB
Save Type: SRAM
CRC1: 67D20729
 
Press Button...

[+] Read ROM

Saving to N64/ROM/SMASH BROTHERS/71/...
[*******************]
CRC32... 04C9D3B1 -> Nintendo All-Star! Dairantou Smash Brothers (Japan).z64
Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Game Cartridge

Searching database...

Title: MARIOKART64
Serial: NKTJ
Revision: 0
ROM Size: 12 MB
Save Type: 4K EEPROM
CRC1: 6BFF4758
 
Press Button...

OSCR HW5 V13.1

[+] Nintendo 64 (3V)

[+] Flash Repro

Clock Generator not found

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

[+] About

Cartridge Reader
github.com/sanni
2023 Version 13.1

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

[+] About

[+] Self Test (3V)

[+] About

[+] Self Test (3V)

Self Test

Remove all Cartridges
before continuing!!!

Press Button...

Testing 1K resistor 
Testing short to GND
Testing short between pins
Testing Clock Generator

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

[+] About

[+] Self Test (3V)

Self Test

Remove all Cartridges
before continuing!!!

Press Button...

Testing 1K resistor 
Testing short to GND
Testing short between pins
Testing Clock Generator

Press Button...

OSCR HW5 V13.1

[+] Self Test (3V)

Self Test

Remove all Cartridges
before continuing!!!

Press Button...

Testing 1K resistor 
Testing short to GND
Testing short between pins
Testing Clock Generator

Press Button...

### System used

N64 Cartridge

### Describe the issue.

I have a problem to option "Reflashing to N64 Repro" the following message appears : "Clock Generator not found"

ptromig, 2024-01-28:
adjunto imagenes
[image]
[image]
[image]
[image]
[image]
[image]

Ancyker, 2024-06-10:
Did you buy it from a seller in China? If so, see: https://github.com/sanni/cartreader/discussions/835#discussioncomment-6770793

sanni, 2024-06-13:
If your Cart Reader has a 2.54mm slot for N64 instead of the correct 2.5mm you need to print out this sleeve for the N64 slot or else you will short out your Cart Reader: https://github.com/sanni/cartreader/blob/a981bb0c3e9143cb1268e77311eadaa620eff621/case/n64sleeve.stl