sanni/cartreader #904: Dumping Sega Cards with the Retron 3-in-1 adapter

flohen95, 2024-02-25:
### OSCR firmware version

13.1

### OSCR hardware version

HW5 Rev3

### System used

Sega Cards

### Describe the issue.

I'm trying to dump a Sega Card (Hang-On in this case) via the Retron 3-in-1 adapter inserted into the Mega Drive Cartridge slot. However, the card (which works using the adapter in an actual Mega Drive) is not detected (the Header info doesn't get read) and selecting a 32 kb ROM size manually results in a bad dump (with a different CRC32 with each attempt).
Is the Retron adapter not supported for Sega Cards or am I doing something wrong? I'm selecting SMS Retron 3in1 as the menu option, of course. Dumping Master System cartridges through the adapter works as well (although I usually use the built in Master System slot), as do GameGear cartridges.
Thanks for the hard work and I hope you can resolve my issue.

PsychoFox11, 2024-02-25:
I believe @PsyK0p4T dumped some of the cards, but wasn't able to get them all to properly dump. He might have some info to share though!

PsyK0p4T, 2024-02-26:
Hi,
At the time of writing, the Retron 3-in-1 adapter is not supported to read SEGA Cards. Current code works with SMS / GG carts  (and MARK III / SG-1000 ones if you use a Retrofreak adapter instead of Retron) but not the SEGA Cards yet.

I think the only way to dump some of them with OSCR is currently by using an official My Card adapter, plugged into a Mark III slot. [picture](https://user-images.githubusercontent.com/87064902/198355949-d04e7476-b531-4197-82a5-2dff7c2222af.jpg
)
With this setup, I've dumped a few EUR and JPN cards. BUT, be warned that it worked for around 50% of the cards I've tested. Many of them are still not well detected.

flohen95, 2024-02-26:
Alright, thanks for the quick response! I was just worried I was doing something obviously wrong - of course, it would be great if support could be implemented eventually, but it's good to know that I wasn’t at fault here or anything.