sanni/cartreader #914: Unable to dump 64MB GBA carts

PsychoFox11, 2024-03-15:
### OSCR firmware version

13.2

### OSCR hardware version

HW5 Rev5

### System used

Select a system

### Describe the issue.

The GameBoy Advance video carts were recently added to the gba.txt file, and these work great for the 32MB carts. However, the 64MB carts don't seem to be able to be dumped.

In the txt file, 64MB is specified as the size, but it shows size as Unkown, and then prompts you to pick a size, with the limit being 32MB.

I tried the only thing I could think of and edited the ino to give a 64MB option on that menu. I got a 64MB dump but it was just the first 32MB repeated. I'm guessing something needs to switch banks.

Is this something that could be looked into for the OSCR?

PsyK0p4T, 2024-03-15:
Can you open the cartridge and take a picture of the pcb ? If it uses Matrix Memory chips, then a bad dump is expected. Some 32 and 64MB video carts are using them.
This mapping is not yet supported and needs to be added to OSCR (Duplo worked on that and his code may be ported ? https://github.com/endrift/duplo)

endrift, 2024-03-16:
Hi, Duplo dev here. The only 64 MB cartridges I know of are the Game Boy Advance Video movies (and maybe one episode pack?), and Duplo can dump all of these. The "mapper" (it works very differently from normal mappers) is pretty simple to set up, so I can help if needed.

PsychoFox11, 2024-03-16:
It's on a Matrix chip which I understand are the more difficult ones to work with, but I hope the OSCR could be tweaked to handle it!
[image]

endrift, 2024-03-16:
Yep, that's the normal type of Matrix 3D Memory cartridge.

PsychoFox11, 2024-03-16:
Nice, I'll check out duplo. Need to get the hardware first, only have DS models with no GB slot currently. Ill work on that!

PsyK0p4T, 2024-03-21:
@endrift , have you maybe tried to port your code to OSCR ? And then Psychofox may help to validate it with his GBA Video carts. I'm grabbing some of them too, but low chance that they contain these Matrix chips inside.

endrift, 2024-03-21:
The trick to finding one with a 64 MB Matrix memory chip is to get one of the movies. Shrek, Shrek 2 and Shrek 2 + Shark Tale always have the chip

Ancyker, 2024-05-07:
> It's on a Matrix chip which I understand are the more difficult ones to work with, but I hope the OSCR could be tweaked to handle it! [...]

The PDF posted here might help with this as it's a similar (but not identical) chip: https://github.com/sanni/cartreader/discussions/942#discussioncomment-9314580