sanni/cartreader #916: Support for Atari 7800 Bankset Bankswitch (needs Adapter update)

smesgr9000, 2024-03-20:
### OSCR firmware version

main 2024-03-24

### OSCR hardware version

HW5 Rev1

### System used

all/none/doesn't matter

### Describe the issue.

I have extended the current code to support Bankset Bankswitch
http://7800.8bitdev.org/index.php/Bankset_Bankswitching

needs some hardware modification on the adapter. Currently @sakman55  7800 adapter has HALT not wired in version 1.00. Thus I used a jumper wire 7800 /HALT (pin 2) to SNES /RST (Pin 26)
I have also added some additional NOPs at write bank switching code. Without switch those 16k was not reliable for me. Those seems to be more stable now for me - also for normal supercard games.
I have successfully dumped Petscii robot with the diff below.

```diff --git a/Cart_Reader/7800.ino b/Cart_Reader/7800.ino
index c8584f0..73078a2 100644
--- a/Cart_Reader/7800.ino
+++ b/Cart_Reader/7800.ino
@@ -44,6 +44,7 @@
 // CONTROL PINS:
 // CLK(PH1) - SNES CPUCLK
 // R/W(PH6) - SNES /RD
+// HALT(PH0) - SNES /RESET
 
 //******************************************
 //  Supported Mappers
@@ -58,9 +59,10 @@ static const byte PROGMEM a7800mapsize[] = {
   4, 4, 4,  // Double Dragon/Rampage 128K [78AC]
   5, 3, 3,  // Realsports Baseball/Tank Command/Tower Toppler/Waterski 64K [78S4]
   6, 3, 3,  // Karateka (PAL) 64K [78S4 Variant]
+  7, 1, 4,  // Bankset switching
 };
 
-byte a7800mapcount = 7;  // (sizeof(a7800mapsize) / sizeof(a7800mapsize[0])) / 3;
+byte a7800mapcount = 8;  // (sizeof(a7800mapsize) / sizeof(a7800mapsize[0])) / 3;
 byte a7800mapselect;
 int a7800index;
 
@@ -96,7 +98,7 @@ void setup_7800() {
   DDRL = 0xFF;
 
   // Set Control Pins to Output
-  //       ---(PH0)   ---(PH3)   ---(PH4)   ---(PH5)   R/W(PH6)
+  //       HALT(PH0)  ---(PH3)   ---(PH4)   ---(PH5)   R/W(PH6)
   DDRH |= (1 << 0) | (1 << 3) | (1 << 4) | (1 << 5) | (1 << 6);
 
   // Set TIME(PJ0) to Output (UNUSED)
@@ -106,7 +108,7 @@ void setup_7800() {
   DDRC = 0x00;
 
   // Setting Control Pins to HIGH
-  //       ---(PH0)    ---(PH3)   ---(PH4)   ---(PH5)   R/W(PH6)
+  //       HALT(PH0)   ---(PH3)   ---(PH4)   ---(PH5)   R/W(PH6)
   PORTH |= (1 << 0) | (1 << 3) | (1 << 4) | (1 << 5) | (1 << 6);
 
   // Set Unused Data Pins (PA0-PA7) to Output
@@ -263,7 +265,32 @@ void writeData_7800(uint32_t addr, uint8_t data) {
   NOP;
   NOP;
   NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
   PORTH |= (1 << 6);  // R/W(PH6) HIGH = READ
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
 
   DDRC = 0x00;  // Reset to Input
 }
@@ -321,20 +348,11 @@ void readROM_7800() {
 
   switch (a7800mapper) {
     case 0:  // Standard 16K/32K/48K [7816/7832/7848]
-      if (a7800size > 1)
-        readSegment_7800(0x4000, 0x8000);  // +16K = 48K
-      if (a7800size > 0)
-        readSegment_7800(0x8000, 0xC000);  // +16K = 32K
-      // Base 16K
-      readSegment_7800(0xC000, 0x10000);  // 16K
+      readStandard_7800();
       break;
 
     case 1:  // SuperGame 128K [78SG]
-      for (int x = 0; x < 7; x++) {
-        writeData_7800(0x8000, x);         // Banks 0-6
-        readSegment_7800(0x8000, 0xC000);  // 16K * 7 = 112K
-      }
-      readSegment_7800(0xC000, 0x10000);  // Bank 7 +16 = 128K
+      readSupergame_7800();
       break;
 
     case 2:                              // SuperGame - Alien Brigade/Crossbow 144K [78S9]
@@ -392,6 +410,24 @@ void readROM_7800() {
         readSegment_7800(0x8000, 0xC000);  // 16K * 4 = 64K
       }
       break;
+
+    case 7: // Bankset switching
+      if (a7800size > 3) {
+        setHalt_7800(1);
+        println_Msg(F("READ Supergame"));
+        display_Update();
+        readSupergame_7800();
+        setHalt_7800(0);
+        println_Msg(F("READ Supergame"));
+        display_Update();
+        readSupergame_7800();
+      } else {
+        setHalt_7800(1);
+        readStandard_7800();
+        setHalt_7800(0);
+        readStandard_7800();
+      }
+      break;
   }
   myFile.close();
 
@@ -405,6 +441,38 @@ void readROM_7800() {
   wait();
 }
 
+void readStandard_7800() {
+  if (a7800size > 1)
+    readSegment_7800(0x4000, 0x8000);  // +16K = 48K
+  if (a7800size > 0)
+    readSegment_7800(0x8000, 0xC000);  // +16K = 32K
+  // Base 16K
+  readSegment_7800(0xC000, 0x10000);  // 16K
+}
+
+void readSupergame_7800() {
+  for (int x = 0; x < 7; x++) {
+    writeData_7800(0x8000, x);         // Banks 0-6
+    readSegment_7800(0x8000, 0xC000);  // 16K * 7 = 112K
+  }
+  readSegment_7800(0xC000, 0x10000);  // Bank 7 +16 = 128K
+}
+
+void setHalt_7800(byte on) {
+  if(on == 1) {
+    PORTH |= (1 << 0);  // HALT(PH0) HIGH = SALLY
+  } else {
+    PORTH &= ~(1 << 0); // HALT(PH0) LOW = MARIA
+  }
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+  NOP;
+}
+
 //******************************************
 // ROM SIZE
 //******************************************
@@ -560,6 +628,8 @@ void checkStatus_7800() {
     println_Msg(F("BASEBALL/ETC [78S4]"));
   else if (a7800mapper == 6)
     println_Msg(F("KARATEKA(PAL)[78S4]"));
+  else if (a7800mapper == 7)
+    println_Msg(F("BANKSET"));
   print_Msg(F("ROM SIZE: "));
   print_Msg(a7800[a7800size]);
   println_Msg(F("K"));
@@ -582,6 +652,8 @@ void checkStatus_7800() {
     Serial.println(F("Realsports Baseball/Tank Command/Tower Toppler/Waterski [78S4]"));
   else if (a7800mapper == 6)
     Serial.println(F("Karateka (PAL) [78S4 Variant]"));
+  else if (a7800mapper == 7)
+    Serial.println(F("Bankset"));
   Serial.print(F("ROM SIZE: "));
   Serial.print(A7800[a7800size]);
   Serial.println(F("K"));
@@ -637,6 +709,8 @@ void setMapper_7800() {
         println_Msg(F("BASEBALL/ETC  [78S4]"));
       else if (a7800mapselect == 6)
         println_Msg(F("KARATEKA(PAL) [78S4]"));
+      else if (a7800mapselect == 7)
+        println_Msg(F("BANKSET"));
       display_Update();
       if (i == (a7800mapcount - 1))
         i = 0;
@@ -665,6 +739,8 @@ void setMapper_7800() {
         println_Msg(F("BASEBALL/ETC  [78S4]"));
       else if (a7800mapselect == 6)
         println_Msg(F("KARATEKA(PAL) [78S4]"));
+      else if (a7800mapselect == 7)
+        println_Msg(F("BANKSET"));
       println_Msg(FS(FSTRING_EMPTY));
 #if defined(ENABLE_OLED)
       print_STR(press_to_change_STR, 1);
@@ -702,6 +778,8 @@ void setMapper_7800() {
         println_Msg(F("BASEBALL/ETC  [78S4]"));
       else if (a7800mapselect == 6)
         println_Msg(F("KARATEKA(PAL) [78S4]"));
+      else if (a7800mapselect == 7)
+        println_Msg(F("BANKSET"));
       println_Msg(FS(FSTRING_EMPTY));
 #if defined(ENABLE_OLED)
       print_STR(press_to_change_STR, 1);
@@ -738,6 +816,8 @@ void setMapper_7800() {
         println_Msg(F("BASEBALL/ETC  [78S4]"));
       else if (a7800mapselect == 6)
         println_Msg(F("KARATEKA(PAL) [78S4]"));
+      else if (a7800mapselect == 7)
+        println_Msg(F("BANKSET"));
       println_Msg(FS(FSTRING_EMPTY));
 #if defined(ENABLE_OLED)
       print_STR(press_to_change_STR, 1);
@@ -770,7 +850,8 @@ setmapper:
   Serial.println(F("4 = Double Dragon/Rampage [78AC]"));
   Serial.println(F("5 = Realsports Baseball/Tank Command/Tower Toppler/Waterski [78S4]"));
   Serial.println(F("6 = Karateka (PAL) [78S4 Variant]"));
-  Serial.print(F("Enter Mapper [0-6]: "));
+  Serial.println(F("7 = Bankset"));
+  Serial.print(F("Enter Mapper [0-7]: "));
   while (Serial.available() == 0) {}
   newmap = Serial.readStringUntil('\n');
   Serial.println(newmap);
```

PsyK0p4T, 2024-03-20:
Hey, thanks for that !
Could you attach the full 7800.ino file please ? and some pics of your adapter once modified, it could help people to do the same :)

sakman55, 2024-03-21:
Updated Gerbers uploaded (UNTESTED):  [Atari 7800 Adapter v0.2.zip](https://github.com/sanni/cartreader/files/14680657/Atari.7800.Adapter.v0.2.zip)

PsychoFox11, 2024-03-21:
I would like a pic - I don't know the first thing about Gerber files, but if someone could explain the modification to me or show pics, I've got a few carts this would likely help

smesgr9000, 2024-03-22:
> Updated Gerbers uploaded (UNTESTED): [Atari 7800 Adapter v0.2.zip](https://github.com/sanni/cartreader/files/14680657/Atari.7800.Adapter.v0.2.zip)

great thanks :)

smesgr9000, 2024-03-22:
> Hey, thanks for that ! Could you attach the full 7800.ino file please ? and some pics of your adapter once modified, it could help people to do the same :)

please either download the file and do in your local repository:
git apply patch.diff
or copy the stuff via button into your clipboard and do, paste clipboard, enter and CTRL+D
git apply -

PsychoFox11, 2024-03-22:
@sakman55 Are you adding this to the ino permanently as well, since you updated the gerbers? Viewing the diff I could make a good guess but worried I'll make a mistake there and don't use git myself currently. Also not sure what version that diff applies to as I'm always on the latest nightly build and sometimes an ino from 2 days ago won't work anymore.

Sharing the full one should just be a drag and drop if it isn't too much to ask, but if it gets merged into the project, that works too!

sakman55, 2024-03-23:
> @sakman55 Are you adding this to the ino permanently as well, since you updated the gerbers? Viewing the diff I could make a good guess but worried I'll make a mistake there and don't use git myself [...]

Sorry, no.  I'm not running the latest version.

smesgr9000, 2024-03-23:
@PsychoFox11 I created a pull request. Maybe that is easier for you? With this I also removed two additional debug lines still left in the above diff.

PsychoFox11, 2024-03-23:
Thanks, I just want to be sure I get it right and haven't used git for actual coding in about a decade!
Hopefully it can be merged in, though I'd still love to see a pic of that bodge. I understand the instructions and can count pin numbers haha, but it is always good to see just to ensure that like we aren't referring to pins somehow differently or anything.

PsyK0p4T, 2024-03-24:
Fix merged ! I don't have retail carts anymore under my hands so... I can't test if it brokes anything, I'm relying on you ^^
Yup as PsychoFox said, a picture of the adapter fix would be welcome :)

smesgr9000, 2024-03-24:
i have tested one 32k, 48k game and one "supergame". The bankset with normal is not tested atm, because afaik there is no cartridge yet.
I close this issue

closed 2024-03-24 by smesgr9000

PsychoFox11, 2024-03-29:
@smesgr9000  I bodged mine and testing with Pescii Robots. Can you tell me what the proper CRC32 you got was? Or the header user to test the dump? I think I'm just not applying to right header info but would like to confirm this is working.

PsychoFox11, 2024-03-30:
@smesgr9000 @sakman55 I'd appreciate some help - sorry for the double tag - I realized my dump is garbage, but also this doesn't make sense. You stated the tested game as Petscii Robots, which is 256kb, but this only allows selection up to 128kb. What option did you choose to dump it with this diff? I still think I made the bodge right but getting garbage dumping at 128kb setting.

PsychoFox11, 2024-03-30:
Ok sorry for all the noise - I got it to work. I think I had a cart seating issue at one point. I found the demo was the same size and used its header which worked great.
Though it was not clear at all that I'd get a 256kb dump by picking 128kb. Does it just autodetect at that point, or are these supposed to be twice size?