sanni/cartreader #917: Reading 256KBit Saves from ExHirom Repro Cart

Cowboyjunkie, 2024-03-23:
### OSCR firmware version

13.1

### OSCR hardware version

HW5 Rev4

### System used

SNES/Super Famicom

### Describe the issue.

The Bahamut Lagoon English Translation by Near is proably the only Cart that has ExHirom and 256KBit SRAM. When trying to read the Save, I only get the first Bank (0xB0) and the rest ist filled with 0xFF.
Trying to Test the SRAM results in ERROR on 0x00 and OK on 0xFF.
Turns out the "romtype == EX" routines are missing the "// Large Sram Fix" the "HI" and "LO" romtypes have. The function stays in the same bank (0xB0) and tries to read the whole 256KBits from the same bank, resulting in "FF" when going over 64KBits.
I copied and pasted the "// Large Sram Fix" to the "EX" part (marking it "// Large EX Sram fix). 
I attached the modified SNES.ino for you.
BTW: Since the reader read "FF" on error it would be better to Test the SRAM with "0x0F" and "0xF0" instead of "0x00" and "0xFF".
Regards,
John
P.S.: the "romtype == SA" section is also missing the large sram fix...

Cowboyjunkie, 2024-03-23:
[SNES.ino.txt](https://github.com/sanni/cartreader/files/14729571/SNES.ino.txt)

sanni, 2024-06-30:
Added your suggestions to the master branch 
https://github.com/sanni/cartreader/commit/d1b1c88af0b3cf2995b7a8ccde052d58ec7793c9

closed 2024-06-30 by sanni