sanni/cartreader #922: SNES/SFC - Yuyu no Quiz de GO!GO! ROM Bank issue?

jrz1333, 2024-04-11:
### OSCR firmware version

13.1

### OSCR hardware version

HW5 Rev5

### Attach OSCR_LOG.txt file here

[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/14940942/OSCR_LOG.txt)

### System used

SNES/Super Famicom

### Describe the issue.

So, I think I may have found an issue with one of the games for SNES/SFC for backing up. I was trying to back up YuYu no Quiz de GO!GO!, and every time I back it up, it seems to have an issue with the CRC32 and the Checksum not matching the database and the initial scan. From doing some quick reading, it looks like the issue lies within the type of ROM type it sees. For example: in the log, you'll see that it's labeled to be HiROM SlowROM. But checking against SuperFamicom.org, this should be LoROM SlowROM. 

Is there a way to change this either within the CFG files or force the type to read LoSlowROMs? It's a low priority issue, But I also wasn't sure if anyone else was experiencing something similar. I couldn't find anything from searching forums either but it could be how I was searching.

PsychoFox11, 2024-04-11:
On the same menu that has Read Rom, you should see Force Cart Type. I've been on 13.2 a while but I'm pretty sure it was there before.

PsychoFox11, 2024-04-11:
If you choose Force Cart Type from the same menu as Read ROM, and choose LoRom, does that not work?

jrz1333, 2024-04-11:
It does not, it reports back something different. What it reports back is as follows:

Checksum.... EEC0!= 3BB0
Invalid Checksum
CRC32...93A8BC26 -> Not Found

sanni, 2024-04-11:
The internal rom name of "Yuyu no Quiz de GO!GO!" is one character too long so it overwrites the byte that the Cart Reader uses to identify lorom or hirom with "!". "!" is 0x21 in hex which is used for hirom games so that's why the OSCR thinks it's a hirom game and dumps incorrectly.

Something similar has happened with Krusty's Super Fun House (U) 1.0 & Contra 3 (U): https://github.com/sanni/cartreader/blob/master/Cart_Reader/SNES.ino#L1003

A special case needs to be added for "Yuyu no Quiz de GO!GO!" something like this:
``
else if (strcmp("3BB0", checksumStr) == 0) { 	// Detect invalid romType byte due to too long ROM name (22 chars)
romType = LO;   				// Yuyu no Quiz de GO!GO!
}
``

jrz1333, 2024-04-12:
Following up on this, I added the lines as follows: 

  romType = snesHeader[0xFFD5 - headerStart];
  if ((romType >> 5) != 1) {  // Detect invalid romType byte due to too long ROM name (22 chars)
  } else if (strcmp("3BB0", checksumStr) == 0) {
    romType = LO;             // LoROM   // Krusty's Super Fun House (U) 1.0, Contra 3 (U), Yuyu no Quiz de GO!GO!
  } else if (romType == 0x35) {
    romType = EX;  // Check if ExHiROM
  } else if (romType == 0x3A) {
    romType = HI;  // Check if SPC7110
  } else {
    romType &= 1;  // Must be LoROM or HiROM
  }
  
  Added the line as intended above the "romType = LO" and just changed the descriptor for the LoROM line to include Yuyu. Just to decrease any added clutter. 

The game backed up just fine and loads in emulators no problem. Thanks for the help, Sanni! I'm not sure if my suggestion earlier made sense or not, it was more of a "driving to work" thought than anything. Maybe it makes sense or is too complicated.

closed 2024-06-13 by sanni