sanni/cartreader #929: Atari 2600 menu broken

plaidpants, 2024-04-20:
### OSCR firmware version

built latest from github and V13.1

### OSCR hardware version

HW3

### Attach OSCR_LOG.txt file here

NA

### System used

Atari 2600

### Describe the issue.

Atari2600 menu does not work, cannot move menu selection dot to next item, present in V13.1 and latest github release. Last working version was V12.8

smesgr9000, 2024-04-27:
I have reworked cart selection today. Could you retest - please update your SD card with the newest files. If not could you describe your issue in more detail. I see no change between 12.8 and 13.1 with menu interaction.

plaidpants, 2024-04-28:
Thanks, I will give it a go tomorrow.

plaidpants, 2024-04-28:
Sill not working, here is a video of both the atari2600 menu not working and the SNES menu working from the same build.

https://github.com/sanni/cartreader/assets/8979271/38e9b746-215b-4ded-8c2b-d0e6e8709adf

https://github.com/sanni/cartreader/assets/8979271/9696f09c-b4f4-462e-ad48-2b591bda2d98

smesgr9000, 2024-04-29:
thats odd. The code handling the menu is the same for SNES and Atari 2600.  Could you please check if all other systems are working. Sadly I don't have the hardware to debug this issue. Another solution to pinpoint the issue would be to https://git-scm.com/docs/git-bisect this issue until the commit can be narrowed down.

sanni, 2024-04-30:
I am not able to recreate this issue with my HW3. Both V13.1 and the current GitHub code work fine. Also the Atari 2600 module was not changed between V12.8 and V13.1, it's the same file.

Have you enabled any other additional modules or options besides Atari 2600? If not try disabling a big module like NES to see if it's a memory issue.

Are you using the portable Arduino IDE from the release section or your own Arduino IDE setup? If you're using your own Arduinio IDE try using the libraries from of the portable release.

plaidpants, 2024-04-30:
Gameboy menu works
**NES menu does not work**
SNES menu works
N64 menu works
Genesis menu works
Sega Master System menu works
**Atari 2600 menu does not work**

plaidpants, 2024-04-30:
[Config.h.txt](https://github.com/sanni/cartreader/files/15172054/Config.h.txt)
Here is the modified config.h file I used.

plaidpants, 2024-04-30:
OK, I think I figured it out, if I turn off ENABLE_VSELECT the atari 2600 menu works, now the question is do I have the ENABLE_VSELECT or not?

PsychoFox11, 2024-04-30:
Not sure what you're asking, but if you have a voltage switch, vselect is off. Vselect is an optional board that sets the voltage for you. If you don't have a voltage switch you probably have one, and can test its functionality with a multimeter, should be 5v on most and 3.3v on GBA/N64.
I have no idea what happens if you configure it incorrectly. I have one unit with it, one without it, but always set it for the specific one.

smesgr9000, 2024-05-01:
@PsychoFox11  could you check with his configuration on your device? I think @sanni is right and has properly something to do with rom used. VOLTAGE_SELECT does require more program memory than without.

PsychoFox11, 2024-05-01:
> @PsychoFox11 could you check with his configuration on your device? I think @sanni is right and has properly something to do with memory used. VOLTAGE_SELECT does require more program memory than without.

@smesgr9000 Sure, though I'm on HW5 - is vSelect a HW3 option? (I haven't really paid attention to pre-HW5 stuff as it is where I came in).

I can try with vSelect enabled when it shouldn't be though, and use the config file but changing the HW version. Will try to look into that tomorrow (well much later today as it is 4am) most likely.

smesgr9000, 2024-05-01:
ah ok sorry thought you had a HW3 - no i don't think this is useful.

PsychoFox11, 2024-05-01:
@smesgr9000 @plaidpants From the page on vSelect - [https://github.com/sanni/cartreader/wiki/Automatic-Voltage-Selection](https://github.com/sanni/cartreader/wiki/Automatic-Voltage-Selection) - it is only an option for HW5, so if they say it works when disabled, and are on HW3, it sounds like working as intended...

PsychoFox11, 2024-05-01:
It does seem like a good update to throw a compile error of some sort if vselect is enabled and HW is <5

Ancyker, 2024-05-01:
> It does seem like a good update to throw a compile error of some sort if vselect is enabled and HW is <5 I'd prefer that over having the code ignore it cause people need to know what they are setting.

A warning would also work, i.e.
```c
#warning VSELECT enabled but is only supported on HW5, disabling.
```

Ancyker, 2024-05-01:
> @smesgr9000 @plaidpants From the page on vSelect - https://github.com/sanni/cartreader/wiki/Automatic-Voltage-Selection - it is only an option for HW5, so if it works when disabled, on HW3, it sounds like working as intended...

There's nothing technically preventing someone from using VSELECT on HW3, it's just that it doesn't make much sense to since you'd still need to switch the clock. I targeted HW5 with it because it was the only switch left on it other than the power switch.

Ancyker, 2024-05-01:
@plaidpants I'm curious, what happens if you update with https://oscr.tools/update ?

Ancyker, 2024-05-01:
> > @plaidpants I'm curious, what happens if you update with https://oscr.tools/update ?
> > Edit: Oh right the Atari menu isn't enabled in that build, well that won't be useful derp 😅
> 
> I am able to update and the NES menu that did not work previously works now however like you said the atari2600 menu is not enabled in that build.

Reset your `Config.h` to untouched and only enable the options you checked in the updater.

Specifically, you probably want `HW3`, `ENABLE_UPDATER`, and `clockgen_installed` (if you have the clock generator). Don't enable anything else. 

Once you do that, see if it still works. If it does, enable the Atari core (`ENABLE_2600`) and see if it continues to work.

Don't touch anything other than those options while testing. Enable things one at a time so that if it breaks you know the last option you enabled is what broke it.

-

The Firmware Updater on OSCR.Tools is a good test because the binaries are cached forever once built -- that means anyone selecting the same options as you is getting literally the exact same binary. This helps rule out a compilation bug specific to your system.

plaidpants, 2024-05-01:
Rebuild with HW3, ENABLE_UPDATER, and clockgen_installed, all menus work, added ENABLE_2600, still works.

smesgr9000, 2024-05-20:
is there a TODO open?

plaidpants, 2024-05-20:
There was a discussion about adding a compile warning or error if the wrong
combinations of defines was encountered. That's all I think was to-do.

Ancyker, 2024-05-20:
Yeah, there's not much to do here. They enabled a hardware option in the firmware that their hardware doesn't have.

Ancyker, 2024-05-20:
Btw, the reason it caused that is VSELECT on HW5 connects to PD7 for toggling the voltages. On HW5 this was the status LED, but on HW3 it's the input button:

[image]

So on HW3, a different pin would need to be chosen for VSELECT's toggle if we were to want to add support for it on HW3.

commit 2b01f7fc (2024-06-02): Add checks for unsafe configurations

* Fixes #929
* Updated `ci.yml` to define `GITHUB_CI`
* Compilation will now error if VSELECT is enabled on hardware versions that don't support it (any version other than HW5).
  * HW4 and SERIAL might work but are untested. The error message for these versions reflects this.
* Erroring when unsafe configurations are defined can be overridden by defining `ALLOW_UNSAFE_CONFIG` which will throw a warning instead.
  * This should only be done if trying to implement or test on a currently unsupported platform.
  * If both `GITHUB_CI` and `ALLOW_UNSAFE_CONFIG` [... cut]
closed 2024-06-02 by Ancyker