sanni/cartreader #939: Atari 2600 F6 mapper issue

plaidpants, 2024-05-01:
### OSCR firmware version

latest

### OSCR hardware version

HW3

### Attach OSCR_LOG.txt file here

N/A

### System used

Atari 2600

### Describe the issue.

This line myFile.write() should be removed as readDataArray_2600() already writes the data to the file. Otherwise you get extra data in the file.
   case 0x04: // F4SC Mapper 32KB \w RAM
    case 0xF4: // F4 Mapper 32KB
      for (int x = 0; x < 8; x++) {
        readData_2600(0x1FF4 + x);
        if(a2600mapper == 0xF4) {
          readSegment_2600(0x1000, 0x1200);
        } else {
          outputFF_2600(0x100); // Skip 0x1000-0x10FF RAM
          readDataArray_2600(0x1100, 0x100);
        }
        readSegment_2600(0x1200, 0x1E00);
        // Split Read of Last 0x200 bytes
        readDataArray_2600(0x1E00, 0x1F4);
        **//myFile.write(sdBuffer, 500);**
        for (int z = 0; z < 12; z++) {
          // Set Bank to ensure 0x1FFC-0x1FFF is correct
          readData_2600(0x1FF4 + x);
          sdBuffer[z] = readData_2600(0x1FF4 + z);
        }
        myFile.write(sdBuffer, 12);
      }

smesgr9000, 2024-05-01:
thanks for the hint. Should be fixed my only 32k game now dumps fine.

plaidpants, 2024-05-01:
I verified this change with my Boulderdash cart and it dumps fine now.

plaidpants, 2024-05-01:
FYI here is a 3E mapper for boulderdash
[2600.ino.txt](https://github.com/sanni/cartreader/files/15179706/2600.ino.txt)

smesgr9000, 2024-05-03:
thanks. I will close this issue than. If I understood your last post correctly you would like to contribute? Please add another issue or provide a pull request - so the issue is not mixed up

closed 2024-05-03 by smesgr9000

plaidpants, 2024-05-04:
pull request created.