sanni/cartreader #944: GB dumping to root of SD card

tjaldridge68, 2024-05-10:
### OSCR firmware version

13.1

### OSCR hardware version

HW5 Rev5

### Attach OSCR_LOG.txt file here

[OSCR_LOG.txt](https://github.com/sanni/cartreader/files/15270436/OSCR_LOG.txt)

### System used

Game Boy (Color)

### Describe the issue.

Game Boy and Game Boy Color files are being placed onto the root of the SD card rather than inside of a dedicated GB folder. Started out with a clean SD card and placed the files onto the root. Dumping a GBA game created a GBA folder but dumping a GB game does not yield the same result.

tjaldridge68, 2024-05-10:
If no GB folder exist on the SD card, it will create a file called GB rather than a folder. Manually creating the directories prior to dumping a GB game results in the games being placed in the correct location.

Ancyker, 2024-05-10:
I tried on latest master and it created the folder fine. Did you format the SD card? What options do you have enabled?

closed 2024-05-10 by tjaldridge68