sanni/cartreader #957: BOM missing?

olDirdey, 2024-06-07:
### OSCR firmware version

asde

### OSCR hardware version

HW1

### Attach OSCR_LOG.txt file here

asd

### System used

Game Boy (Color)

### Describe the issue.

where is the BOM for the project?

Ancyker, 2024-06-07:
https://github.com/sanni/cartreader/wiki/What-to-order

olDirdey, 2024-06-07:
I don't see any bom there either, just a much too small picture of an aliexpress shopping cart

Ancyker, 2024-06-07:
Scroll down, there are links to everything.

olDirdey, 2024-06-07:
isn't there just a part number so that I can search the parts from a proper source? I don´t like to order stuff from ali

Ancyker, 2024-06-07:
For most things? No. For some things? Yes.

* Get the [Clock Generator from Adafruit](https://www.adafruit.com/product/2045) (or [here](https://store.starshade.dev/product/oscr-and-accessories/oscr-accessories/rtc-clkgen-module)).
* Bulk Capacitor: [PLS1A471MDO1TD](https://www.digikey.com/en/products/detail/nichicon/PLS1A471MDO1TD/4991343)
* 1K Resistor: [PR01000101001JR500](https://www.digikey.com/en/products/detail/vishay-beyschlag-draloric-bc-components/PR01000101001JR500/595993)
* CIC: [PIC12F629-I/SN](https://www.digikey.com/en/products/detail/microchip-technology/PIC12F629-I-SN/459170) (or [here](https://store.starshade.dev/product/oscr-hw5-cic))

The rest either isn't available from a reputable source or is so simple it's not worth the effort (i.e. pin headers).