sanni/cartreader #965: N64 saves

capnsbum10, 2024-06-14:
### OSCR firmware version

13.1

### OSCR hardware version

HW3

### System used

N64 Cartridge

### Describe the issue.

I’m trying to read the save data to copy to a duplicate game cartridge. When I go to read save data it says error no data received. Ive made sure the clock was defined and had clock 1 switch on

sanni, 2024-06-14:
Can you open the N64 cartridge an post an image of the small 8-pin eeprom chip? Especially what is written on it. Then I can look if I have one with the same markings to check it out on my side.

You can also try [V10.2](https://github.com/sanni/cartreader/releases/tag/V10.2] which uses the old eeprom dump code. (CLK switch set to on.)
Or since you are on HW3 [V7.2](https://github.com/sanni/cartreader/wiki/Cart-Reader-V3#older-releasesfor-testing-newer-releases-will-work-with-v3-hardware-too) which uses an even older code that needs the CLK switch set to **off**.

capnsbum10, 2024-06-15:
Waiting on screwdrivers to open the carts. I downgraded to version 10.2 and now it says reading eeprom… but hasn’t done anything else, appears to be frozen there.

capnsbum10, 2024-06-15:
Tried to go down to firmware 7.3 but system warned of low memory availability and stability problems. Now the cart reader won’t turn on

sanni, 2024-06-17:
In options.h you need to change:

````
//******************************************
// CHOOSE HARDWARE VERSION
//******************************************
#define HW4
//#define HW3
````
to :
````
//******************************************
// CHOOSE HARDWARE VERSION
//******************************************
//#define HW4
#define HW3
````

So that it runs on HW3.