sanni/cartreader #966: N64 Carts show "Gamepak Error" when trying to read

my2k2zx2, 2024-06-14:
### OSCR firmware version

13.1

### OSCR hardware version

HW5 Rev3

### Attach OSCR_LOG.txt file here

[OSCR_LOG.txt](https://github.com/user-attachments/files/15844754/OSCR_LOG.txt)

### System used

N64 Cartridge

### Describe the issue.

I bought a prebuilt reader, it was delivered today.  I've been able to read SNES, Gameboy and Genesis without any issue.  I've now tried N64 but get "Gamepak Error" with any cart I've tried.  I have the n64.txt file on the sd card.  I'm not sure if there is anything more that I'm missing for these carts to read properly.

cmcm711, 2024-06-14:
Can you link to where you bought your OSCR? There are several overseas (China) sellers that are offering improperly-built devices with the wrong parts to unsuspecting buyers.

my2k2zx2, 2024-06-14:
It was from from Aliexpress.  Some of the soldering doesnt look the greatest, probably part of the problem.

cmcm711, 2024-06-15:
So, the AliExpress ones use a Sega Master System cartridge slot (2.54mm pitch) instead of an N64 slot (2.5mm pitch), which will cause a short to your N64 cartridge. You _might_ be able to install a 3D printed bracket to help center the cartridge in the slot, but it's not guaranteed to help. A lot of units have also been reported to have cold solder joints or other faulty components.

my2k2zx2, 2024-06-15:
Do you have a link to the bracket?  I can print one and see if anything changes.

cmcm711, 2024-06-15:
https://github.com/sanni/cartreader/blob/master/hardware/six_slot_adapter/n64sleeve.stl

my2k2zx2, 2024-06-15:
Nope, even with the sleeve it still gives the "gamepak error".  I have the option for a free return so I'm going that route.  Will get a diy kit once the refund is done.