sanni/cartreader #973: purpose of snes.ino "boolean altconf"?

smesgr9000, 2024-06-29:
### OSCR firmware version

V13.3

### OSCR hardware version

HW5 Rev6

### System used

SNES/Super Famicom

### Describe the issue.

What is the purpose of the flag "altconf" in snes.ino?
It seems it only toggles print lines between those variants:
"Rom Size: "
"ROM Size: "

It would like to remove this flag if it hasn't any other purpose.

sanni, 2024-06-30:
I remember I created that variable to signal that the rom header of the SNES cartridge was wrong and instead the config from the snes.txt database was used. 
````
"Rom Size: "
"ROM Size: "
````
was an indicator that this happened before I added the log file where you can now easily read that it got changed.

So yeah, you can remove it as it doesn't have a purpose anymore.

smesgr9000, 2024-07-01:
thanks for the explanation

closed 2024-07-01 by smesgr9000