sanni/cartreader #986: U1 is missing on the Schematic

Helder1981, 2024-07-15:
### OSCR firmware version

N/A

### OSCR hardware version

HW5 Rev6

### System used

all/none/doesn't matter

### Describe the issue.

The PCB file has the U1 part but on the schematic it is not there, can this be fixed?

Helder1981, 2024-07-16:
Sorry I am new to kicad and did not know how to go to sheet 2 that has the U1 there.

Ancyker, 2024-07-16:
Yeah, if you click VSELECT on the main schematic it will open it. There should also be a nav bar on the left I believe. The next revision actually has 5 pages total :3

[image]

Helder1981, 2024-07-18:
It was the V-Select IC. I'm adding the clock generator to the main pcb and maybe the n64 controller port and Super CIC. Planning to just have the cart pcb the 7in1 to just have cart slots.

Helder1981, 2024-07-18:
Ah nice , I've been wanting to include the Mega since I first learned of this project. What are the FFC sockets for?

Ancyker, 2024-07-18:
To connect to the new top PCB which also includes a header for the RetroSix GG slot.

closed 2024-07-27 by sanni