sanni/cartreader #991: Progress bar when erasing Flashrom

StanMOI, 2024-07-18:
### OSCR firmware version

14

### OSCR hardware version

HW5 Rev6

### System used

all/none/doesn't matter

### Describe the issue.

Hello,

Will be nice if we can see the "Progress Bar" when Flashrom are erasing.
Why not to have an option to activate/desactivate "progress bar" in CONFIG file.

Chomemel, 2024-07-18:
It's a super idea. 
Also, would it be possible, to add the diode blinking while READING when you are using the FLASHROM programmer (exactly like reading on a cartridge)? And would it be possible to add it while ERASING and BLANKCHECKING?

sanni, 2024-07-19:
I can only add blinking to the erase function as it's doing a chip erase instead of a sector erase and with chip erase I don't know anything about progress, just done or not done. 
I did however update blank check, verify and read functions with both blinking and progress bar.

StanMOI, 2024-07-19:
Also, 
Before "erasing" starting it's will be nice to have some "confirm erasing" button.
Why not for reading "reading" function also.

closed 2024-07-27 by sanni