sanni/cartreader #995: Support for Evercade

TopHatHipster, 2024-07-23:
### OSCR firmware version

14

### OSCR hardware version

HW3 Alter

### System used

all/none/doesn't matter

### Describe the issue.

Would it be possible to add Evercade cartridge support at some point?

Doing a bit of searching, it should be possible to read Evercade cartridges for backing up and possibly restoring corrupted ones.

- The Evercade cartridge looks like to be a "[microSD card flash soldered onto a GBA cartridge](http://retrotechr.blogspot.com/2020/06/evercade.html?m=1)" with the layout being known as:
    - Pin 1 - DAT2
    - Pin 2 - DAT3
    - Pin 3 - CMD
    - Pin 4 - VDD
    - Pin 5 - CLK
    - Pin 6 - GND
    - Pin 7 - DAT0
    - Pin 8 - DAT1
- The same article as above also points out that the cartridge is formatted as FAT16 - which should be possible with the [SdFat library existing out there for Arduino devices](https://github.com/greiman/SdFat). I'm unsure if this is used for regular SD card support on the cartreader?
- [No-Intro seemingly suggests dd'ing cartridges](https://forum.no-intro.org/viewtopic.php?t=5248) for proper back up.

Would adding this as a software update feature be feasible, with the pre-existing GB/GBA slot present in cartreader builds?

Can't make hardware myself nor am I knowledgeable in Arduino software, but I'd love to help out testing if possible.

PsychoFox11, 2024-07-24:
I only know what I've learned from this project, but if speeds are comparable to current carts it would be a SLOW dump on those 512MB ones for sure. I can just about go for dinner during a 128MB ROM dump lol.

PsychoFox11, 2024-07-24:
Also, looks like pin 1 is used for power on GBA, and ping 4 on Everdrive. Software can move some pins but not the voltage/power, so it would likely need an adapter at the least. At that point, why not just make an evercade to SD adapter and skip the additional hardware?

TopHatHipster, 2024-07-24:
> At that point, why not just make an evercade to SD adapter and skip the additional hardware?

As stated in my issue - I can't make hardware myself hence my suggestion to make it a possible software feature.

However, thanks for the clarification why it wouldn't be possible with another adapter due to the power pins. Will close the issue, due to this blocking problem. 👍

closed 2024-07-24 by TopHatHipster

nsx0r, 2024-07-24:
@smesgr9000 made a cool and simple adapter that uses a GameBoy slot and plugs directly to a microSD port: https://github.com/smesgr9000/microcade
However, no 2 Evercade cartridges are identical so there's no way to compare/verify 2 dumps of the same game.

TopHatHipster, 2024-07-24:
> However, no 2 Evercade cartridges are identical so there's no way to compare/verify 2 dumps of the same game.

It's more so about the files used by said cartridges. AFAIK they should be exactly the same and are checksum checked by the OS present on Everdrive devices.

nsx0r, 2024-07-24:
> > However, no 2 Evercade cartridges are identical so there's no way to compare/verify 2 dumps of the same game.
> 
> It's more so about the files used by said cartridges. AFAIK they should be exactly the same and are checksum checked by the OS present on Everdrive devices.

the files are identical, yes, I'm talking about image (dumps) of the entire cartridge/media, which the OSCR does for cartridges. So even if the OSCR could backup Evercade cartridges, there would be no easy way to automatically identify the cartridge and set the proper settings for dumping.

smesgr9000, 2024-07-24:
yes and no
yes the files are mostly identical. BUT because save states, statistics and _UPDATES_ to the cartridge are stored on card you need to track which files are actually the ROM for every cartridge. There are reports where updates to the cart have bricked the cards file structure and aren't detected by Evercade anymore. Plus an update could also replace ROMs of course.  After all it is just a SD-Card in odd housing.
To be honest I would bother supporting this on cardreader if you could just use your ordinary PC.