sanni/cartreader #996: genesis batman forever and turrican cannot be dumped properly

fakkuyuu, 2024-07-24:
### OSCR firmware version

14.0

### OSCR hardware version

HW3

### Attach OSCR_LOG.txt file here

Cart Info
 
Name: BATMANFOREVER
bramCheck: 0100
Size: 32 MBit
ChkS: BC5F
Sram: None
 
Press Button...

[+] Read ROM

Saving to MD/ROM/BATMANFOREVER/2036/...
[*******************]
Internal checksum...Error
2C50

CRC32... E7E3958C -> Not found
Press Button...

OSCR HW3 V14.0

[+] Mega Drive/Genesis

[+] Game Cartridge

Cart Info
 
Name: BATMANFOREVER
bramCheck: 0100
Size: 32 MBit
ChkS: BC5F
Sram: None
 
Press Button...

[+] Read ROM

Saving to MD/ROM/BATMANFOREVER/2037/...
[*******************]
Internal checksum...Error
3087

CRC32... BD7C2F94 -> Not found
Press Button...

OSCR HW3 V14.0

[+] Mega Drive/Genesis

[+] Game Cartridge

Cart Info
 
Name: BATMANFOREVER
bramCheck: 0100
Size: 32 MBit
ChkS: BC5F
Sram: None
 
Press Button...

[+] Read ROM

Saving to MD/ROM/BATMANFOREVER/2038/...
[*******************]
Internal checksum...Error
B35B

CRC32... 173FB5FC -> Not found
Press Button...

OSCR HW3 V14.0

[+] Mega Drive/Genesis

[+] Game Cartridge

Cart Info
 
Name: TRIA
bramCheck: A501
Size: 4 MBit
ChkS: A509
Sram: None
 
Press Button...

[+] Force ROM size

ROM Size: 1 Mbit

Press left to Change
and right to Select

ROM Size: 2 Mbit

Press left to Change
and right to Select

ROM Size: 4 Mbit

Press left to Change
and right to Select

ROM Size: 8 Mbit

Press left to Change
and right to Select

ROM Size: 12 Mbit

Press left to Change
and right to Select

ROM Size: 16 Mbit

Press left to Change
and right to Select

ROM Size: 20 Mbit

Press left to Change
and right to Select

ROM Size: 24 Mbit

Press left to Change
and right to Select

ROM Size: 32 Mbit

Press left to Change
and right to Select

ROM Size: 40 Mbit

Press left to Change
and right to Select

ROM Size: 1 Mbit

Press left to Change
and right to Select

ROM Size: 2 Mbit

Press left to Change
and right to Select
ROM Size: 2 Mbit
Press Button...

[+] Read ROM

Saving to MD/ROM/TRIA/2039/...
[*******************]
Internal checksum...Error
0825

CRC32... 3102E091 -> Not found
Press Button...

OSCR HW3 V14.0

[+] Mega Drive/Genesis

[+] Game Cartridge

Cart Info
 
Name: TRIA
bramCheck: E501
Size: 4 MBit
ChkS: E509
Sram: None
 
Press Button...

[+] Force ROM size

ROM Size: 1 Mbit

Press left to Change
and right to Select

ROM Size: 2 Mbit

Press left to Change
and right to Select

ROM Size: 4 Mbit

Press left to Change
and right to Select

ROM Size: 8 Mbit

Press left to Change
and right to Select

ROM Size: 12 Mbit

Press left to Change
and right to Select

ROM Size: 16 Mbit

Press left to Change
and right to Select

ROM Size: 20 Mbit

Press left to Change
and right to Select

ROM Size: 24 Mbit

Press left to Change
and right to Select

ROM Size: 32 Mbit

Press left to Change
and right to Select

ROM Size: 40 Mbit

Press left to Change
and right to Select

ROM Size: 1 Mbit

Press left to Change
and right to Select

ROM Size: 2 Mbit

Press left to Change
and right to Select
ROM Size: 2 Mbit
Press Button...

[+] Read ROM

Saving to MD/ROM/TRIA/2040/...
[*******************]
Internal checksum...Error
0825

CRC32... 25F70BA1 -> Not found
Press Button...

OSCR HW3 V14.0

[+] Mega Drive/Genesis

[+] Game Cartridge

Cart Info
 
Name: TRIA
bramCheck: A501
Size: 4 MBit
ChkS: A509
Sram: None
 
Press Button...

### System used

Sega Mega Drive/Genesis

### Describe the issue.

I don't know why

fakkuyuu, 2024-07-24:
thanks.
I can't check the operation on the actual machine because my room is a mess right now. I'll check it later.

PsychoFox11, 2024-07-24:
I did confirm that Batman Forever should at least get a known checksum. No cart to test, but the known version has a valid checksum as opposed to Turrican.

fakkuyuu, 2024-07-24:
thanks.I forgot to post a photo.One thing the two cartridges have in common is that they have a small number of pins, so I thought that was related.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

PsychoFox11, 2024-08-01:
All I can say is my Turrican looks just like that and dumped fine!

fakkuyuu, 2024-08-01:
thank you. I found a set of screwdriver bits in my room that can open these cartridges, so I'll check what's inside.

fakkuyuu, 2024-08-01:
There was no apparent rust or disconnection.
[image]
[image]

PsychoFox11, 2024-08-01:
The pins on the front look a little sketch to me but it could be the lighting. Just like there could be stuff preventing good contact. I recommend the 99% alcohol clean with q-tips until they stay white, and maybe going over it with a soft rubber pencil eraser (not one of the harder ones) to loosen stuff and do it again. That's all I really got though, and don't have that one to test myself. I double checked and I've only dumped 'Forever' on SNES.

PsychoFox11, 2024-08-01:
Ok I'm always curious about stuff, my local game store has Batman Forever for $7 so I had them hold a copy, and will try to get it tomorrow to test on my side. I'll let yo know how it goes.

fakkuyuu, 2024-08-01:
thanks. We have confirmed that there are three types of MD SLOTs sold on ALIEXPRESS.
A type:WINGTAT is laser engraved.
B type: No engraving. Looser insertion feeling than A
C type: Pin shaped like SNES and 64. ears on the sides

I created a cartreader using A and B and tried it, but the result was the same

fakkuyuu, 2024-08-01:
I tried cleaning the terminals using absolute ethanol, but it didn't work.
I'm hesitant to cut out the stickers on the screw holes of the turrican.
[image]

fakkuyuu, 2024-08-01:
I tried cleaning the terminals using absolute ethanol, but it didn't work.
I'm hesitant to cut out the stickers on the screw holes of the turrican.
[image]
[image]

PsychoFox11, 2024-08-01:
Ah you mean the rental sticker? I do like to keep those on but I will punch through them for PCB pics haha.
Lemme see how Batman Forever is for me, and if I can dig up my copy of Turrican I can try it again.
How many Genesis carts have you dumped successfully? Just wondering if it could be an issue with a pin on your OSCR that some games don't use etc.

fakkuyuu, 2024-08-01:
I think I've dumped more than 50 cartridges. Among them, I remember that MEGAGAMES10 was also quite difficult to dump, but I managed to extract it by inserting it slightly to the right. This time, I tried adjusting the insertion in a similar way, but it didn't work

PsychoFox11, 2024-08-01:
And other carts still dump fine? You may have mentioned but just making sure.

fakkuyuu, 2024-08-01:
I tried with the cassette I had on hand earlier. It worked well.
[image]
[image]

PsychoFox11, 2024-08-01:
My big difference is I'm on HW5 but from what I understand that shouldn't change what dumps here. I grabbed the game today and will be looking into it in a bit after work.

PsychoFox11, 2024-08-02:
I'm having very similar issues with Batman. Turrican still dumps fine. Batman has 1 more pin than Turrican so I don't think it is about the pin count, but I'm gonna open an issue. I'll tag you and probably use your PCB pic!

PsychoFox11, 2024-08-02:
Ok, posted the issue with some additional info. Seems it is reading the first 3MB, which is the game, just fine. The last MB of padding isn't being read right though. I'd think it wasn't mapped to the edge connector, but someone dumped it with a Retrode, so I'm a bit lost, and posted what I could see.
The ROM DOES work just fine - is that your experience with Batman as well?

fakkuyuu, 2024-08-02:
It worked with kegafusion. I've never tried it

PsychoFox11, 2024-08-02:
Yeah so the game is fine, it is just the padding that follows it being read incorrectly. If all you care about is running the game you're good. I'm about accurate chip-content preservation though!

fakkuyuu, 2024-08-02:
I also opened turrican. The wiring on the back of the board was uneven and I thought it was corroded, but when I checked it with a tester, it seemed to be connected. I haven't checked all the wiring though. When I searched for the model number of the board, I found something that seemed to be the reason why dumping was not possible.
[image]
[image]
https://x.com/mikepavone/status/1564459841430495233

PsychoFox11, 2024-08-02:
That's probably it - I do wonder if the OSCR could be tweaked to dump that mapper - of course, emulators would need to support it as well. It does seem like tracking down and desoldering/dumping that chip might be worth it. Even if not supported by emulation, I like preservation, but I bet someone could emulate what's going on there.
I opened my Turrican and it is a single chip very similar to the Batman board, so that's gotta be it. Mine must have been after the, shall we say, boo-boo.
Might have to take a chance with a couple loose carts on eBay. Seems to be worth about $20. I have it fully CIB which seems to be a lot more (included ads and all).

PsychoFox11, 2024-08-02:
@sanni This one got a bit long but there is a variant of Turrican that doesn't dump properly. To save you a trip to Twitter, here's the explanation - but I'm wondering if a desolder is gonna be the only way to copy the ROM, or the the OSCR could adapt to this mapper. Any thoughts? Just tagging you cause I wrote it off in my other post about Batman, but now I see there's a difference, as mine is just a ROM, and this one has extra logic chips.
[image]