sanni/cartreader #997: 26L6420 flashing settings

alx5962, 2024-07-24:
### OSCR firmware version

14

### OSCR hardware version

HW3

### System used

Flashrom Programmer

### Describe the issue.

I'd like to flash a 26L6420 (SOP44) eprom using this cart reader but I can't find infos about this topic.

I already own the 26L6420 PCB adapter to flash them from : https://github.com/sanni/cartreader/tree/bd3eaa106b6e4cbdffbfa89a3fb5fedf3e028b17/pcb/adapters. It's shaped as a SNES adapter and it's not mentioned in the wiki.
I'm using the latest firmware on the reader and enabled FLASH16 in the firmware. 

First, I'd like to know how to set the 4 switches and what options do I have to use in the Flashrom programmer menu. I guess it's the 16 bits Flash one (but when I select it, it only displays [+] 16 bit flash).

Many thanks for this great project! I'm creating my own Gameboy game and I use it a lot!

sanni, 2024-07-24:
Seems like you found a bug. 😄 
Please update to the latest Github code and try again.

Switches like this:
[image]    
And "16bit Flash" menu selection.

alx5962, 2024-07-24:
Thank you for the fast reply and for the very fast fix! 

So I just flashed the new firmware and now I have this error: "Flash ID 1010 unknown flashrom". In the flash.ino file:

if (flashid == 0xC2FC) {
    println_Msg(F("MX26L6420 detected"));

So I added my own flashid, now let's try if all works well.

alx5962, 2024-07-27:
Some feedback: I used another eprom and now flashid is correctly detected as 0xC2FC. I tried to erase it, 20 mn later still erasing it. How long does it take to erase, usually? I will double check every pins as I use an adapter like this:

[image]

So I had to wire myself every pins to the SNES/Eprom adapter.

alx5962, 2024-07-27:
I double checked and my wiring is fine. Flashid is 0xC2FC and blankcheck works fine but after 1 hour erasing still erasing (led off). Also tried to flash a blank eprom and after 1 hour I have the same result and no blinking (led always on).

I have a question about the 26l6420 adapter schematics found at:  https://github.com/sanni/cartreader/blob/bd3eaa106b6e4cbdffbfa89a3fb5fedf3e028b17/pcb/adapters/26L6420_adapter_schematics.png

[image]

Somes pins have no direct link with the chip numbers: A21 is connected to WE, WE to BYTE, A20 to WP but some have the correct number: A18 is connected to A18, A17 to A17,... On the offical doc (below), A21 is an address input so it should be linked to A21.

Do these changes are linked to eprom flashing with this adapter? Or a issue with the schematic?

* Official MX26L6420 datasheet:

[image]

[image]

sanni, 2024-07-27:
I don't remember who made this adapter but from a quick look at it you can ignore the pin descriptions. They seem to just have forgotten to change the symbol in KiCad.

[image]

alx5962, 2024-07-27:
Thank you for the feedback.  Any idea about why the led is not blinking while flashing? Or is it only in 8 bit mode?

sanni, 2024-07-27:
One thing you could try is flashing V4.3 from the [firmware archive](https://github.com/sanni/cartreader/wiki/Firmware-Archive), it should be the first version with 26l6420 support, to check if maybe over the past 5 years something got changed that broke  26l6420 support.

herzmx, 2024-07-27:
I try this and works, but 16bit Flash adaptador need a mod to work.
First cut trace on PH1
[image]
And next join OE to PH3
[image]

I don't remember if firmware need a mod, but i take a look in my old files.

This testes un HW5

alx5962, 2024-07-28:
> I try this and works, but 16bit Flash adaptador need a mod to work. First cut trace on PH1 ![20240727_141727](https://private-user-images.githubusercontent.com/5693135/352775797-2eb4d43e-fd6a-4624-bbc6-74cf7abff844.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIxNTA5MzQsIm5iZiI6MTcyMjE1MDYzNCwicGF0aCI6Ii81NjkzMTM1LzM1Mjc3NTc5Ny0yZWI0ZDQzZS1mZDZhLTQ2MjQtYmJjNi03NGNmN2FiZmY4NDQuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjhUMDcxMDM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTFiMTcxNDVkNGUzOGRiZTg3NmY0MmEwZDE4NmFjMTI1MWI2MmI3ODA2MTNlNjg4MmI1ZGQzMWY1ODg1YzljNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.3gr0fpB3AIaUyzpwYAWZR-vpttNlVBrrY42ft7yuDqY) And next join OE to PH3 ![20240727_141744](https://private-user-images.githubusercontent.com/5693135/352775929-5727dc13-673a-472c-b1ab-8cccfaaf45a7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIxNTA5MzQsIm5iZiI6MTcyMjE1MDYzNCwicGF0aCI6Ii81NjkzMTM1LzM1Mjc3NTkyOS01NzI3ZGMxMy02NzNhLTQ3MmMtYjFhYi04Y2NjZmFhZjQ1YTcuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjhUMDcxMDM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Njk4ZjRhYzY4YWE2MzhjMzAxOWJlMDM1ZDE1ZDYwYjA2NmM0M2RiYWEzY2U2NzAzZjAwYzEyZWM5MzY1YWY4OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.lrJHwLUAvj6kYF1mpnWCeQSUBxFXf-rtQjL4E0ha9Ks)
> 
> I don't remember if firmware need a mod, but i take a look in my old files.
> 
> This tested on HW5

A big thank you for your hardware update! I will fix my adapter later today. Does this mod update the flashid number or is it still C2FC ? And can you tell me what firmware version is flashed on your reader?

alx5962, 2024-07-28:
@herzmx I followed your instructions to mod my adapter:

[image]

I reverted to the latest firmware and checked again my wiring. Now the detected flashid is 9090 and not C2FC. Don't know if it's ok or if my tests fried the eprom...

Also I noticed on your picture another wire on ground. Not related to this mod?

herzmx, 2024-07-30:
I'm back, i lose my old files, but with rewind in my mind, so when 16-BIT adapter released PH1 is tracked to CPU CLK in SNES, now in HW5 PH1 don't tracked to SNES Slot, because use CICCLK istead.
The change in adapter is to use PH3 to activate/deactivate OE and read words, this change works editing `word readWord_Flash` function
```C
word readWord_Flash(unsigned long myAddress) {
  PORTF = myAddress & 0xFF;
  PORTK = (myAddress >> 8) & 0xFF;
  PORTL = (myAddress >> 16) & 0xFF;

  // Arduino running at 16Mhz -> one nop = 62.5ns
  __asm__("nop\n\t");

  // Setting OE(PH3) LOW
  PORTH &= ~(1 << 3);

  __asm__("nop\n\t"
          "nop\n\t"
          "nop\n\t"
          "nop\n\t"
          "nop\n\t"
          "nop\n\t");

  // Read
  word tempWord = ((PINA & 0xFF) << 8) | (PINC & 0xFF);

  __asm__("nop\n\t");

  // Setting OE(PH3) HIGH
  PORTH |= (1 << 3);
  __asm__("nop\n\t"
          "nop\n\t"
          "nop\n\t"
          "nop\n\t"
          "nop\n\t"
          "nop\n\t");

  return tempWord;
}
```

Please test...
[FLASH.zip](https://github.com/user-attachments/files/16422126/FLASH.zip)

I don't know if this affect other flash memorys, so i don't make PR at this time, enjoy your DG-ROM conversion 👍

alx5962, 2024-07-30:
@herzmx thank you for your reply. I'm using HW3 so your update will work on it too or is it specific to HW5?

So you noticed my message on arcade projects forum 😁

herzmx, 2024-07-30:
Sorry i didn't see HW version on issue description, so no HW mod it's need because CLK1 switch in HW3 activates PH1, but with HW mod you need use the FLASH.ini attached in previous message.

Please test, it should work on HW3 as well, in my previous test I had issues with the memory having a sticker with chinese characters.

alx5962, 2024-07-30:
> Sorry i didn't see HW version on issue description, so no HW mod it's need because CLK1 switch in HW3 activates PH1, but with HW mod you need use the FLASH.ini attached in previous message.
>
> Please [...]

no problem at all!

I flashed the reader using the flash.ino file you provided. Now flashid is well detected (C2FC). But erasing doesn't seem to work (or takes more than 1 hour).  I gonna try to find another 26l6420 eprom to check if the current one is not dead.

TailsnicRetroworks, 2024-11-28:
Hi. I have seen with this project, 26L6420 eproms can be flashed, so I want to ask if this type of rom is fully compatible with erasing/writting/reading mode for making the entire PCB, many thanks in advance

Daze909, 2025-05-16:
This was extremely helpful as I wanted to use it to write to the WonderSwan.
Even with the current version (v14.5), it is necessary to change the word readWord_Flash.
I've left here the Gerber for the PCB I made for myself.

[Gerber_cartreader_26L6420-Adapter.zip](https://github.com/user-attachments/files/20241294/Gerber_cartreader_26L6420-Adapter.zip)