sanni/cartreader #1001: SNES: can't fully dump Retro-Bit/JNNEX multi-rom repros

jd213, 2024-07-29:
### OSCR firmware version

14.0

### OSCR hardware version

HW5 Rev3

### Attach OSCR_LOG.txt file here

OSCR HW5 V14.0

[+] Super Nintendo/SFC

[+] SNES/SFC cartridge

Searching database...
Checksum: 35FE

Name: WOODI SFC TEST
Revision: 0
Type: LoROM SlowROM
ICs: ROM ONLY
ROM Size: 256 KB (8 banks)
Save Size: 32 KB
Checksum: 35FE
Press Button...

[+] Force cart type

[+] Use header info

Press Button...

[+] Force cart type

[+] 4MB LoROM 256K SRAM

Press Button...

[+] Read ROM

Saving to SNES/ROM/LoROM/509/...
[*******************]
Checksum... 1CA4 != 35FE
Invalid Checksum
CRC32... 90C65DD7 -> Not found

### System used

SNES/Super Famicom

### Describe the issue.

Currently, it's possible to dump just part of the multi-game Retro-Bit SFC repros (such as "16-bit Collection Athena Vol. 1") if one Forces the cart type to "4MB LoROM 256K SRAM" and then edit the dump with a hex editor as shown in the responses to #822 , but it doesn't dump enough of the data to get all of the roms unfortunately.

Would it be possible to add a mode that can dump the full cart?

I believe these carts are still fairly cheap (and available new at some locations) in Japan, but I can send one for testing purposes if someone in the US wants to take a look at it.

(GitHub isn't letting me add the "Feature Request" tag for some reason, sorry)

sanni, 2024-07-30:
Can you try what happens if you set the ROM type to LoROM with 8MB?
In [SNES.ino line 456](https://github.com/sanni/cartreader/blob/master/Cart_Reader/SNES.ino#L456) change 128 to 256 then select "4MB LoROM 256K SRAM" (which now should be 8MB)

jd213, 2024-07-31:
Ah, thanks for the suggestion.
Unfortunately it looks like the same incomplete romset is repeated in the 2nd 4MB.
Got the same behavior with both Retro-Bit's Athena Vol.1 and Jaleco Vol.1.

sanni, 2024-08-02:
One more thing you can try is updating to the newest Github master branch code and then try to dump it using the Flash Repro menu under SNES.

jd213, 2024-08-03:
Hmm, I installed the newest master and tried dumping with 8MB flashsize, LoROM (and a couple tries as 6MB or ExLoROM), and each of the various flashtypes, but they also resulted in incomplete/repeated romsets.  

Here's a pic of the chips if it helps, at first glance I thought it was the same as the Joe & Mac cart but looks like the U2 chip is different: 
[image]

sanni, 2024-08-04:
Maybe cut out the menu ROM and load it into an emulator and then check the debugger what happens when e.g. the last game is selected. Maybe the CPLD acts as a more advanced mapper compared to the simple 74HC139 we find in the typical Chinese repros and the menu outputs a command to switch to the second half of the flash chip.

jd213, 2024-08-04:
Thanks, I opened the menu rom in bsnes plus and tried to look for clues, but unfortunately I'm mostly just a solder monkey and didn't really know what to look for.  I appreciate the suggestion, though.