sanni/cartreader #1013: Arcadia dumping incorrectly

PsychoFox11, 2024-08-13:
### OSCR firmware version

14.0

### OSCR hardware version

HW5 Rev5

### System used

Emerson Arcadia 2001

### Describe the issue.

Only one cart has been tested so far, but I hope to get more, and see an issue on this one. There is only one BIT incorrect in the dump. My AtariMax Maxflash dumper dumps it correctly, matching the very old hash on No-Intro. OSCR just records a single bit wrong.

Here's a diff with the problem shown in red. Maxflash/No-intro on top, OSCR on bottom. As I hope to do more, and it's a fairly neglected system, I'd love to see this fixed!
[image]

sanni, 2024-08-14:
Try adding a few additional NOP; [here](https://github.com/sanni/cartreader/blob/master/Cart_Reader/ARC.ino#L131) and see if that changes anything.

breyell, 2024-08-15:
> Try adding a few additional NOP; [here](https://github.com/sanni/cartreader/blob/master/Cart_Reader/ARC.ino#L131) and see if that changes anything.

Added a single NOP and that resolved the issue - thanks!

PsychoFox11, 2024-08-15:
Nice! (Just noting I filed this based on Bonzo's dump as he's making my adapter and borrowed my cart). So should be good!

PsyK0p4T, 2024-08-15:
added that to code :) Thanks !

closed 2024-08-15 by PsyK0p4T