sanni/cartreader #1018: Can't find #define in Cart_Reader.ino

superbonaci, 2024-08-19:
### OSCR firmware version

2021 7.3

### OSCR hardware version

HW3

### Attach OSCR_LOG.txt file here

not needed

### System used

Game Boy (Color)

### Describe the issue.

I have the Sanni Cart Reader v3, and trying to compile the latest version, following this guide: https://github.com/sanni/cartreader/wiki/How-to-flash-the-Arduino

Step 7 says this:

7) Next you need to go to the config.h tab(earlier version have this in Cart_Reader.ino) and define your hardware version by removing the // in front of your version. You can also remove // before #define enable_XXX to enable add-on modules you need.

The `Cart_Reader.ino` doesn't have the define as listed in the screenshot:

[image]

sanni, 2024-08-19:
I have updated the wiki to represent the current state

[image]

superbonaci, 2024-08-19:
Thanks. Is possible to backup the current firmware or Sketch that is already in the cart reader? Because my 7.3 is currently working and just in case the new build has some issue.

sanni, 2024-08-20:
Yes that is possible, I have just now added a BACKUP.bat file to the current V14.3 portable release. It launches the included avrdude.exe to create a backup.hex (tested in Windows 10 & 11).

Here is the used command in case you are on another OS or do not use the portable release (edit path and COM port to your system)
````
"C:\Arduino IDE\hardware\tools\avr\bin\avrdude.exe" -C "C:\Arduino IDE\hardware\tools\avr\etc\avrdude.conf" -c wiring -P COM3 -b 115200 -p m2560 -U flash:r:backup.hex
````

closed 2024-08-29 by sanni