sanni/cartreader #1028: Can't dump Starwing (SNES, PAL)

BenoitCaron, 2024-09-16:
### OSCR firmware version

14.3

### OSCR hardware version

HW5 Rev5

### Attach OSCR_LOG.txt file here

OSCR HW5 V14.3

[+] Super Nintendo/SFC

[+] SNES/SFC cartridge

Searching database...
Checksum: 00FF
Header CRC32: AD9F2E43
Header CRC32: 546FF471
Header CRC32: A6932322
Header CRC32: B00AC917
Header CRC32: CB040CF6
Header CRC32: 86DAEEC6
Header CRC32: A5F599F1
Header CRC32: 5D1AFD5E
Header CRC32: 5101B422
Header CRC32: 5D9EED2B
Found

Name: STAR WING
Revision: 0
Type: LoROM SlowROM
ICs: ROM SuperFX
ROM Size: 1 MB (32 banks)
Save Size: 32 KB
Checksum: 00FF
Press Button...

[+] Read ROM

Saving to SNES/ROM/STAR WING/6/...
[*******************]
Checksum... 0214 != 00FF
Invalid Checksum
CRC32... C66B320D -> Not found
Press Button...

[+] Read ROM

Saving to SNES/ROM/STAR WING/7/...
[*******************]
Checksum... 02A6 != 00FF
Invalid Checksum
CRC32... 177D37D6 -> Not found

### System used

SNES/Super Famicom

### Describe the issue.

I have been trying to dump my PAL Starwing cartridges for ages now, expecting it would magically work with every new firmware release. I've thoroughly cleaned the contacts and inspected the PCB, but can't find anything wrong with it.
[image]
[image]

PsyK0p4T, 2024-09-16:
Oh... epoxy-style ! Is this coming from a PAL cartridge ?
Various Star Wing carts I've dumped until now were all using big PCBs, and dumped fine with OSCR. I've never tested one like yours.
[image]
At least you got a rev0, confirmed by header and PCB label, so final CRC32 should be 865F1A71.

Have you tried to double clean the pins ? With a soft rubber for example. Also, using 5V ?

BenoitCaron, 2024-09-16:
Yes the epoxy had me wondering at first, but there seems to be several revisions of Starwing, so it seems my cartridge is legit. I have cleaned the pins to the best of my ability, and I do use 5V. Again, this is the only cartridge I haven't been able to dump so far (tested every slot with every cartridge type multiple times).

sanni, 2024-09-20:
Is there a way to tell what version of the PCB is inside the cartridge? Does the label differ from the normal version? It looks interesting.

BenoitCaron, 2024-09-20:
Absolutely, here it is
[image]