sanni/cartreader #1038: Can't figure out why checksum is invalid

Crusa1der, 2024-10-24:
### OSCR firmware version

2023 Version 13.1

### OSCR hardware version

Select hardware version

### System used

SNES/Super Famicom

### Describe the issue.

Hey, I'm new to emulation in general and have been trying to get Super Metroid to work but keep getting the message of an invalid checksum. 

Title: Superman Metroid
Revision: 0
Type: LoROM FastROM
IC's: ROM SAVE
Rom size: 3MB (96 banks)
Save Size: 8 KB
Checksum: F8DF
Press button...

Read ROM

Saving to SNES/ROM/Super Metroid/5/...
[**************]
Checksum... 76F0 != F8DF
Invalid Checksum
CRC32...493C9489 -> Not found
Press button...

It will then save a rom labeled Super Metroid.sfc as well as a text file of the same name and I've tried running it on both higan (which loads the game but doesn't output video or sound [shows fps and game title]), and Snes9x (which instantly tells me it's an invalid checksum). Any help would be appreciated. Also, I have not tried with any other cartridge, if that helps.

PsychoFox11, 2024-10-24:
Do you get a consistent CRC32 on multiple dumps? I haven't checked this one and doubt a first party title would have an invalid checksum, but some games do - the console doesn't care. But the fact that it isn't running right in emulators tells me the dump is bad.
Make sure it is clean as heck with some 99% alcohol and q-tips, till they are white. The OSCR is far pickier than real consoles about dirt.
If you aren't getting a consistent CRC32 on each dump something is definitely not right with the connection (though in some cases even if it is consistent there could be a connection issue, but when it isn't, there definitely is).

Crusa1der, 2024-10-24:
I cleaned the cart until there was no black or discoloration on the q-tip and tried again, got the same exact invalid checksum 
CRC32...493C9489 -> Not found
Also, sorry for the late reply, didnt have any rubbing alcohol.

PsychoFox11, 2024-10-24:
Are you sure it is a legit cart? Got a pic of the PCB? Even a bootleg with wrong CRC32 should load in an emulator though. Maybe with a checksum warning, but it would work.

Crusa1der, 2024-10-24:
[image]

Crusa1der, 2024-10-24:
> I just noticed the "Superman Metroid" - something isn't right there

No, it does say super metroid. My autocorrect was on. 
[image]

Ancyker, 2024-10-24:
Are you on the Discord?

Crusa1der, 2024-10-24:
> Are you on the Discord?

No, I don't really use those kinds of apps. Let me try to resend the photo
[image]

PsychoFox11, 2024-10-25:
Other SNES games dump ok? Especially any with similar settings?

Crusa1der, 2024-10-25:
> Other SNES games dump ok? Especially any with similar settings?

I have not tried other snes or sfc games because I dont have others, but I can get another sometime next week or so.

PsychoFox11, 2024-10-26:
Yeah you really need more than one to confirm if something with the dumper may be wrong. Ideally a few to test. Could be bad cart, bad OSCR, bad anything at this point.

Barawer, 2024-10-26:
If you are using a computer port or something similar to power the OSCR try using a good wall usb charger. I've had issues like this due to not enough power.

PsychoFox11, 2024-10-26:
Yeah I agree checking things like that, plus running the self check in the menu if you haven't, would be a good start - if other's don't dump well. Do carts for any console dump properly?

Crusa1der, 2024-10-29:
I just got another game to try, sorry for the wait.
Still invalid checksum, but it seems diffrent
[image]

Crusa1der, 2024-10-29:
[image]

Crusa1der, 2024-10-29:
I also cleaned the cart before trying so that can't be it.

PsychoFox11, 2024-10-29:
I'd guess the problem is in your OSCR somewhere, rather than the cart. Does a cart for a different system dump? Unless the problem is on one or more pins that only SNES uses, it would typically affect other consoles.