sanni/cartreader #1044: Six-Slot PCB Text error

ccfman2004, 2024-11-27:
### OSCR firmware version

N/A

### OSCR hardware version

HW5 Rev6

### System used

all/none/doesn't matter

### Describe the issue.

The text on the Six-Slot adapter says "2.5mm 50-Pin N64 Cartridge Slot". Shouldn't it say 2.54mm as the pin spacing was changed a while ago.

nsx0r, 2024-11-27:
The N64 slot does have a 2.5mm pitch, not 2.54mm (that's for the Sega Master System).

closed 2024-11-27 by nsx0r

ccfman2004, 2024-11-27:
I feel like an idiot. I completely misread the change log. I didn't see the word "controller" in "Change N64 controller connector pitch to 2.54mm" for some reason. I hate it when my brain does that.

Sorry about that.