sanni/cartreader #1058: Can't dump XEVIOUS (Famicom)

Kampfer0080, 2024-12-29:
### OSCR firmware version

14.3

### OSCR hardware version

HW5 Rev5

### Attach OSCR_LOG.txt file here

OSCR HW5 V14.3

[+] Famicom

Searching database
for 0FFC433A...

CRC not found in database
Using manual selection
Select Mapping from List

NES CART READER

CURRENT SETTINGS

MAPPER:   0
PRG SIZE: 32K
CHR SIZE: 8K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/Xevious/8/...
[*******************]
CRC32... 67FD72CA -> Not found

Press Button...

OSCR HW5 V14.3

[+] Famicom

Searching database
for 0FFC433A...

CRC not found in database
Using manual selection
Select Mapping from List

NES CART READER

CURRENT SETTINGS

MAPPER:   0
PRG SIZE: 32K
CHR SIZE: 8K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/Xevious/9/...
[*******************]
CRC32... 67FD72CA -> Not found

Press Button...

OSCR HW5 V14.3

[+] Famicom

Searching database
for 0FFC433A...

CRC not found in database
Using manual selection
Select Mapping from List

NES CART READER

CURRENT SETTINGS

MAPPER:   0
PRG SIZE: 32K
CHR SIZE: 8K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/Xevious/10/...
[*******************]
CRC32... 67FD72CA -> Not found

Press Button...

### System used

NES/Famicom/Family Basic

### Describe the issue.

I attempted to dump the XEVIOUS (Famicom) cartridge but was unsuccessful. A "CRC32 not found" message appeared, and the dumped ROM would not boot in an emulator.

Since all my other Famicom cartridges dump correctly, I believe my setup is properly assembled. Suspecting the issue might be specific to this cartridge, I purchased two additional XEVIOUS (Famicom) cartridges for testing, but none of them could be dumped either.

However, all three XEVIOUS cartridges play without issue on an actual Famicom console.

I disassembled the cartridges, verified the continuity of the pins, and cleaned the connectors, but these efforts did not resolve the problem.

[image]
[image]
[image]

Kampfer0080, 2024-12-29:
When viewing the dumped file in a hex editor, the data from address 00000010 to 00008000 was filled with 6F.
[image]
[image]

nsx0r, 2024-12-29:
Hi, I ordered a Xevious FC cartridge, I'll run some tests when I receive it.

splash5, 2025-01-09:
This is because the ROM chip used by XEVIOS needs a high to low transition to latch a new address.
In `read_prg_byte` function, even `set_romsel` will set /ROMSEL signal by A15, when dumping PRG ROM,
A15 is always high (read 0x8000 - 0xffff) and /ROMSEL never goes high, so the new address is never correctly latched by ROM chip.

On real hardware /ROMSEL is also controlled by PHI2 so /ROMSEL will go high when PHI2 goes low.

Kampfer0080, 2025-01-09:
Hi @splash5,  I was able to dump Xevious using the following code you posted in #1059.

static unsigned char read_prg_byte(unsigned int address) {
  MODE_READ;
  PRG_READ;
  ROMSEL_HI;  // add this line
  set_address(address);
  PHI2_HI;
  set_romsel(address);
  _delay_us(1);
  return PINK;
}

Kampfer0080, 2025-01-11:
Hi @splash5, If the above changes are applied to the code, it becomes possible to dump Xevious, but is there a possibility that other Famicom cartridges can no longer be dumped?

For reference, I tested the modified code with 24 Famicom cartridges I own, and I was able to dump all of them without any issues.

splash5, 2025-01-11:
It should not affect other games dumping because this modification makes reading waveform close to real hardware.
But more testing is welcome!

closed 2025-01-16 by sanni