sanni/cartreader #1059: Cant dump NES Tetris (USA) ver

JakubWaikat, 2024-12-30:
### OSCR firmware version

14.3

### OSCR hardware version

HW3

### Attach OSCR_LOG.txt file here

OSCR HW3 V14.3

[+] NES/Famicom

No data found.
Using manual selection
Select Mapping from List

NES CART READER

CURRENT SETTINGS

MAPPER:   1
PRG SIZE: 32K
CHR SIZE: 16K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/Tetris/12/...
[*******************]
CRC32... FE55AE10 -> Not found

Press Button...

### System used

NES/Famicom/Family Basic

### Describe the issue.

When dumping NES Tetris cartridge data (ROM), the cartridge is not recognized (which may be fine), so I select the Tetris version from the menu. Then it fails after dumping with CRC32... FE55AE10 -> Not found.  

The cart itself is working (verified on original hardware afterwards), the dumped file looks weird and does not work in an emulator.    
[image]  

All my other carts (Zelda 1,2, Super Mario, ...) were dumped successfully, so I think my hardware is working fine. Its weird that especially tetris is not working ... I thought that this game has been dumped a thousand times.  

Also the cart looks genuine to me:  
[image]  
[image]
[image]

nsx0r, 2024-12-31:
Hi, this should be fixed now https://github.com/sanni/cartreader/blob/2100cd55f60e8d6e411469f082990edf50dacdc4/Cart_Reader/NES.ino#L1853

JakubWaikat, 2024-12-31:
Hi, thanks for the very fast response. I have tested the fix, but it unfortunately did not work.  

```
OSCR HW3 V14.5

[+] NES/Famicom

No data found.
Using manual selection
Select Mapping from List

NES CART READER

CURRENT SETTINGS

MAPPER:   1
PRG SIZE: 32K
CHR SIZE: 16K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/Tetris/19/...
[*******************]
CRC32... 892B4EC0 -> Not found

Press Button...
```

Is there anything I can do (send you some data or smth) to support you with debugging?

gemarcano, 2024-12-31:
Does Tetris not have a signature that can be picked up on by the OSCR? I find it odd you have to specify its information. With another SEROM cart, Dr. Mario, the firmware correctly identified the cart (even before the fix, I just couldn't dump it).

If you can analyze the dumped game, that can help to figure out what's going on. Specifically you'll want to look for repetitions of large blocks of data usually at some address power of two.

If I under the mapper for Tetris, your game dump should be structured as:
 - 16 bytes of header
 - 32 * 1024 (0x8000) bytes of PRG ROM
 - 16 * 1024 (0x4000) bytes of CHR ROM

Pre-fix, my PRG section was repeating every 16 * 1024 bytes (or, after removing the first 16 bytes, the first 0x4000 bytes were the same as the next 0x4000 bytes).

JakubWaikat, 2024-12-31:
I put the dump into HxD, there is only FF until 0x8000 and also the remainder does not look right IMO. Attached the .nes file:  
[Tetris.zip](https://github.com/user-attachments/files/18283718/Tetris.zip)

gemarcano, 2024-12-31:
Interesting. I can try to take a look at the ROM, but if there's only FF until 0x8000 that sounds like a problem reading from PRG ROM. Even pre-fix at least one of my PRG pages looked right.

gemarcano, 2024-12-31:
We might need help from someone that knows more about the NES than me, because it sounds to me like some control signals are being ignored by your PRG (bad chip), or they're just not making it to the ROM (broken traces on cart, problems with socket, bad traces on adapter board).

Does your OSCR dump other NES games? And does Tetris work on a NES?

JakubWaikat, 2025-01-01:
Yes this is the weird thing ... the OSCR dumps other NES games well (Zelda 1,2; Super Mario Bros + Duckhunt, Mega Man 3; ...) without any issues and the NES cart works in the original hardware without any problems

gemarcano, 2025-01-01:
That's really odd. If other games are dumping, that most likely means that at least the normal control signals for the PRG ROM work. And honestly that should be all that's needed for a SEROM PCB cart, I think. And if Tetris works on the original cart, clearly the NES is reading the PRG ROM.

Hm. This is a wild guess, but if you have a way to check the voltage on the cart while it's plugged in, I wonder if your cart doesn't like that the voltage might be drooping below 4.7V? At least on my OSCR the Vcc on the cart was around 4.6V. Although in _theory_ the chips of that generation should work with voltages down to 4.5V.

Sadly I'm out of ideas. If this is something you want to continue investigating I'd hop on the OSCR Discord and see if there's someone there that might know more.

JakubWaikat, 2025-01-02:
I can probably try that tomorrow, otherwise I will ask in the discord. Thanks for the support!

JakubWaikat, 2025-01-05:
I took a measurement between pins 1 and 36 and got something between 4.640V to 4.720V but never going below 4.6V. I will try to consult discord.

splash5, 2025-01-07:
Could you try modify
`#define MODE_READ \
  { \
    PORTK = 0xFF; \
    DDRK = 0; \
  }`
to
`#define MODE_READ \
  { \
    PORTK = 0x00; \
    DDRK = 0; \
  }`
in NES.ino and try dump it again?

pandidand, 2025-01-07:
Thanks for the suggestion, I tried, but did not work, see the dump attached.  
[image]  
[Tetris.zip](https://github.com/user-attachments/files/18337434/Tetris.zip)

splash5, 2025-01-08:
Ok, so this means PRG ROM is never been activated.
Could you try add one line in read_prg_byte function and try again?
<code>
static unsigned char read_prg_byte(unsigned int address) {
  MODE_READ;
  PRG_READ;
  ROMSEL_HI;  // add this line
  set_address(address);
  PHI2_HI;
  set_romsel(address);
  _delay_us(1);
  return PINK;
}
</code>

pandidand, 2025-01-08:
[image]  

I tried with `MODE_READ \ { \ PORTK = 0x00` and `MODE_READ \ { \ PORTK = 0xFF`, but unfortunatelly both did not work:  
[image]  
[image]

splash5, 2025-01-09:
Hmm that's strange. At least the CHR ROM should be able to dump without problem, because CHR ROM enable signal
is pretty simple.
I also tried wiring my MMC1 dev cart like SEROM and it dumped correctly, this means cartreader itself is able to dump SEROM.

Maybe you can try to check/clean cartridge connector on cartreader and cartridge's gold finger and try it again.

pandidand, 2025-01-14:
OK I checked the contacts anyways although I knew they were fine (since all my carts are in a good condition and I clean the contacts regularly), and I noticed that it may be the case, that my NES connector (of the OSCR) does not reach the contacts on the back side of the PCB.
See images here:
Front side:
[image]
Back side:
[image]

I have then inserted the cart into my NES adapter:
[image]
[image]

The fingers of the adapter are not going up that much, so I took a multimeter and checked the contacts and indeed they were not connecting. I have then managed with some strange angled insertion method to get the contacts connected (I assume barely) and then the dumping was working.

I have then verified with the latest main branch if the current code does dump that cart and it did also work.

So thanks everyone :) 

I will close the issue now.

closed 2025-01-14 by pandidand