sanni/cartreader #1060: NES: Unable to dump Dr. Mario

gemarcano, 2024-12-31:
### OSCR firmware version

commit fb8835b6c411b7a80cbc78025133b5f92ac47dc2

### OSCR hardware version

HW5 Rev5

### Attach OSCR_LOG.txt file here

OSCR HW5 V14.5

[+] NES/Famicom

Searching database
for E5B20B43 or 04718301...

NES CART READER

CURRENT SETTINGS

MAPPER:   1
PRG SIZE: 32K
CHR SIZE: 32K
RAM SIZE: 0K

Press Button...

[+] Read iNES Rom

Saving to NES/ROM/DrMario/78/...
[*******************]
CRC32... 5E7DBC48 -> Not found

Press Button...

### System used

NES/Famicom/Family Basic

### Describe the issue.

Cart_Reader currently assumes that carts with `MMC1` mappers, well, map PRG ROM pages, but that's not the case for SEROM PCBs apparently: https://www.nesdev.org/wiki/MMC1_pinout#SEROM,_SHROM,_SH1ROM

Dr. Mario is a `NES-SEROM-04` PCB cart with a `MMC1B2` mapper. I can confirm from visual inspection that the PRG address lines leaving the mapper are not connected to anything. The CHR address lines are.

This page has more info on the cart, and the PCB images match what I see on mine: https://nescartdb.com/profile/view/516/dr-mario

Trying to dump the game results in the first 0x4000 bytes being repeated twice (effectively, the first "bank" of PRG is repeated twice). CHR seems fine.

I don't see a way to dump this cart at the moment, short of me hacking the MMC1 code just for this cart for the time being.

gemarcano, 2024-12-31:
A fix may be possible to implement once https://github.com/sanni/cartreader/issues/442 is fixed, as SEROM MMC1 mappers are already documented as a submapper type of MMC1: https://www.nesdev.org/wiki/NES_2.0_submappers#001:_5_Fixed_PRG

nsx0r, 2024-12-31:
Hi, this should be fixed now https://github.com/sanni/cartreader/blob/2100cd55f60e8d6e411469f082990edf50dacdc4/Cart_Reader/NES.ino#L1853

gemarcano, 2024-12-31:
At a minimum this does allow Dr. Mario to be dumped correctly. I am not super familiar with NES mappers, but the implementation of the fix looks good, I think it should work for both SEROM et al. carts and regular MMC1 that only have 32K of PRG. Thanks!

closed 2024-12-31 by gemarcano