sanni/cartreader #1061: hw5 and cloak generator with 7800

smesgr9000, 2025-01-01:
### OSCR firmware version

V14.0+

### OSCR hardware version

HW5 Rev1

### System used

all/none/doesn't matter

### Describe the issue.

I had for some time the suspicion the cloak of HW5 interfere with bank switching dumps on Atari 7800 and made those dumps somewhat unreliable.
With "Bounty Bob Strikes Back!" I'm now relatively certain. Does somebody have HW3/HW5 cartreader with either Bob or Bentley Bears (AtariAge or Atari Interactive release) cartridge and can dump the current development branch.

After that please split those dumps into 16kb chuncks for example with linux command split:
`split --bytes=16k ATARI.a78 ATARI_split_`
and report the resulting splits crc32 checksum for example with:
`crc32 ATARI_split_*`

```
diff --git a/Cart_Reader/7800.ino b/Cart_Reader/7800.ino
index 7f53449..898146d 100644
--- a/Cart_Reader/7800.ino
+++ b/Cart_Reader/7800.ino
@@ -134,6 +134,9 @@ void setup_7800() {
   // Start outputting Eeprom clock
   clockgen.output_enable(SI5351_CLK1, 1);  // Eeprom clock
 
+  clockgen.set_clock_disable(SI5351_CLK1, si5351_clock_disable::SI5351_CLK_DISABLE_HIGH);
+  clockgen.output_enable(SI5351_CLK1, 0);
+
   // Wait for clock generator
   clockgen.update_status();
```

PsychoFox11, 2025-01-01:
Is this a newer issue? I dumped Bentley Bear in the past and it matched the digital from AtariAge's forums, which you can see here:
https://datomatic.no-intro.org/index.php?page=show_record&s=74&n=0143

That was back in April. If this is a newer issue I'll have to wait until I can get to the cart again, but could try. I DID have to use the revised version of the 7800 adapter. I had the original and added a bodge wire to make it match the updated version.

smesgr9000, 2025-01-03:
thats interessting. Maybe we discussing two issues here.
1)  Cloak gen - Afiak we currently have a race condition if the cloak generator is present. Most 7800 have a flipflop to switch banks which require Write, A15 AND CLK pin to be on. If the cloak gen CLK is currently in its low phase the bank switch will not work.
Thus the workaround to disable the CLK and set it on HIGH. This may fail the Activison stuff which does do something with the CLK pin.
Why this was more reproducible on this modern releases - I can't say. Maybe the newer flipflop had other timings than the old 80th Hitachi counterpart.

2) with which mapper did you have dumped the AA release? With Alien Brigade/Crossbow? This does indeed do not work for Atari release. Afaik the AA and Atari does use totally different PCBs. The AA has likely a POKEY/HOKEY DIP40. The Atari very likely the same PCB as Bounty Bob with a fully integrated HOKEY (or HOKEY like) ARM-Coprocessor and 512kB Flash. I was hoping at least the rest of the logic is identical. Would be good to know if AA can be dumped like Atari release.
Why do you needed the HALT connected makes no sense to me. Either the AA release uses HALT for something different OR you have triggered the timing issue from above - hard to tell without seeing the PCB.

PsychoFox11, 2025-01-04:
I'll need to dig up the cart again and see. I found my dump folder on my SD but apparently it doesn't record a log file for 7800, I forgot about that. So I'll need to try again when I can. Though fwiw I do see that I dumped it multiple times successfully.