sanni/cartreader #1079: HW V3 can use an N64 controller to move v elox in the menu?

bonna79, 2025-03-09:
### OSCR firmware version

14 +

### OSCR hardware version

HW3

### Attach OSCR_LOG.txt file here

How can I use an N64 controller to be able to interact in the menu and select the games to be dunpare, such as Nes games? 
Thank you

### System used

N64 Controller Test

### Describe the issue.

How can I use an N64 controller to be able to interact in the menu and select the games to be dunpare, such as Nes games? 
Thank you

sanni, 2025-03-10:
Using an N64 controller to navigate the menu has not been implemented. The controller test is just for testing if the N64 controller works.

bonna79, 2025-03-17:
> Using an N64 controller to navigate the menu has not been implemented. The controller test is just for testing if the N64 controller works.

Ah ok. Thank you .

But would it be possible to add the function to move in the menus, through the n64 pad?
Because when I have to dump a game that starts with the letter M, I have to scroll through all the game titles from A to M, and it takes a long time. Or are there other shortcuts to get to the M quickly? Thank you

gemarcano, 2025-03-17:
I don't see any reason it wouldn't be possible to do, other than increasing code size. Could be #define gated. Main issue is having multiple inputs work at the same time-- easiest would be to have controller or buttons. Of course, someone would have to do it. I'm tempted, but I don't have a lot of time right now (I'm actually in the middle of a very slow rework of the N64 controller test code, just to see if I can simplify it).