sanni/cartreader #1082: Dumping Nes Cartige pal/Ntsc

bonna79, 2025-03-30:
### OSCR firmware version

14.5

### OSCR hardware version

HW3 Alter

### Attach OSCR_LOG.txt file here

[OSCR_LOG.txt](https://github.com/user-attachments/files/19525027/OSCR_LOG.txt)

### System used

NES/Famicom/Family Basic

### Describe the issue.

I wanted to know if it was possible to dump recent cartridges that use different mappers like the game, Changeable guardian Estique. Reading and asking in the nesdev forum, someone told me that he uses a mapper 484, which I do not currently believe is supported. I tried to copy the PRG and the CHR, but it did not split anything on the micro sd. The PCB in question has 2 256k PRGs one for the pal and the other for the NTSC version. So how to read the PRG1, PRG2 and CHR separately? To then merge them and create a single .nes file with a correct header? Currently what I get is a nes file that results has a green screen. 
Thank you

nsx0r, 2025-03-30:
Before adding that mapper to the OSCR, the script needs to be tested (I don't own this cartridge).
If you want to help and have some time for testing, feel free to join the Discord channel.

bonna79, 2025-03-30:
> Before adding that mapper to the OSCR, the script needs to be tested (I don't own this cartridge).
> If you want to help and have some time for testing, feel free to join the Discord channel.

Very willingly, even if I understand little, very willingly.

Thank you .